Ultimaker/Cura #2949: Infill Mesh

novio8, 2017-12-10:
Infill Mesh is Broken. 
Cura 2.7 is working, 3.1 is broken

Regards
Mario

smartavionics, 2017-12-10:
Hi, how is it broken, it seems to be working for me:
[image]

novio8, 2017-12-10:
I attached Image, there is no infill, no Top Layer
[image]

smartavionics, 2017-12-10:
You need to make the mesh type specific to that object (special modes -> mesh type), have you done that?

novio8, 2017-12-10:
second image using Infill Mesh
[image]

smartavionics, 2017-12-10:
OK, I can see that is infill mesh.

smartavionics, 2017-12-10:
That model that you have selected to be infill mesh, is solid, isn't it? If it was hollow then it won't be filled with infill.

novio8, 2017-12-10:
Am I doing something wrong ?

novio8, 2017-12-10:
It is a solid no naked edges, no non man-fold edges

smartavionics, 2017-12-10:
Just noticed that you have some very weird shell lines here:

[image]

Look at those red lines that converge towards the top right of the image, their not right!

smartavionics, 2017-12-10:
Hah!, you have spiralize enabled????

smartavionics, 2017-12-10:
No problem, please close the issue when you are happy there is not a real problem, cheers!

closed 2017-12-10 by novio8