Ultimaker/Cura #2950: OS X High Sierra : Crash at startup

bistory, 2017-12-10:
Hi,
I just upgraded Cura from 3.0.4 to 3.1 (not using beta before) and I'm unable to start Cura. Here is the relevant log : 

```
2017-12-10 10:14:47,060 - INFO - FirmwareUpdateChecker.FirmwareUpdateChecker._onContainerAdded [52]: You have a 'Anycubic i3 Mega' in printer list. Let's check the firmware!
2017-12-10 10:14:47,066 - DEBUG - UM.Settings.ContainerRegistry.load [231]: Loading data into container registry took 34.684818983078 seconds
2017-12-10 10:14:47,461 - DEBUG - UM.Controller.setActiveView [86]: Setting active view to SolidView
2017-12-10 10:14:47,547 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-12-10 10:14:47,549 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2017-12-10 10:14:47,551 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
2017-12-10 10:14:47,553 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
2017-12-10 10:14:47,554 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
2017-12-10 10:14:47,556 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
2017-12-10 10:14:47,558 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/env/3.1/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/__startup__.py", line 12, in <module>
2017-12-10 10:14:47,560 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
2017-12-10 10:14:47,562 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
2017-12-10 10:14:47,563 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
2017-12-10 10:14:47,565 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
2017-12-10 10:14:47,567 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/env/3.1/inst/lib/python3.5/site-packages/cx_Freeze/initscripts/Console.py", line 21, in <module>
2017-12-10 10:14:47,569 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 969, in _find_and_load
2017-12-10 10:14:47,571 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
2017-12-10 10:14:47,572 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
2017-12-10 10:14:47,574 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
2017-12-10 10:14:47,576 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/bin/cura_app.py", line 83, in <module>
2017-12-10 10:14:47,578 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 681, in run
2017-12-10 10:14:47,580 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 107, in __init__
2017-12-10 10:14:47,581 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 733, in getMachineManager
2017-12-10 10:14:47,583 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/Settings/MachineManager.py", line 1234, in createMachineManager
2017-12-10 10:14:47,585 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/Settings/MachineManager.py", line 109, in __init__
2017-12-10 10:14:47,587 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/Settings/MachineManager.py", line 357, in setActiveMachine
2017-12-10 10:14:47,588 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/UM/Application.py", line 156, in setGlobalContainerStack
2017-12-10 10:14:47,590 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 212, in emit
2017-12-10 10:14:47,592 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/UM/Signal.py", line 315, in __performEmit
2017-12-10 10:14:47,594 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 500, in _onStackChanged
2017-12-10 10:14:47,596 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 597, in _updateDisallowedAreas
2017-12-10 10:14:47,597 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/Users/ultimaker/build/3.1/build/inst/lib/python3.5/site-packages/cura/BuildVolume.py", line 956, in _getEdgeDisallowedSize
2017-12-10 10:14:47,599 - CRITICAL - cura.CrashHandler.__init__ [65]: ValueError: max() arg is an empty sequence
```

I hope this helps :)

Thanks !

ChrisTerBeke, 2017-12-10:
@LipuFei is this related to the fix you did on Friday?

ChrisTerBeke, 2017-12-10:
Related to https://github.com/Ultimaker/Cura/issues/2798?

ChrisTerBeke, 2017-12-10:
I'm closing this issue in favor of the already existing https://github.com/Ultimaker/Cura/issues/2798. Please keep an eye on that one to track the status of fixing this.

closed 2017-12-10 by ChrisTerBeke