Ultimaker/Cura #2952: Cura 3.1

Thisismydigitalself, 2017-12-10:
I could have sit quietly in the corner and say nothing but i feel like saying that Cura 3.1 released version should not be available for download on Ultimaker website. too many issues.

ChrisTerBeke, 2017-12-10:
The issue are mainly focused on people that have used the beta, as there were a few bugs in the beta that caused project files and the upgrade script results to be stored incorrectly. So far we haven't been able to reproduce on clean installs, both internally and with beta users that have removed their data before running the stable version.

We do extensive QA and system testing before any release, so if bugs would affect a majority of users or use cases it wouldn't have been released in the first place.

We always try to release the most stable Cura we can, but with such complex software there can always be bugs.

Thisismydigitalself, 2017-12-10:
I really want to use 3.1 and you are right - i did install beta versions but uninstalled these and kept only 3.0.4 and 3.1 release . i guess this did not solve my issues with 3.1. now i only have 3.0.4 installed. all other versions were removed/uninstalled.
 
Can you please direct us what should we do for cleaning our PC's from all previous installations so we cab enjoy 3.1?

ChrisTerBeke, 2017-12-10:
There is a Wiki page for user preferences/settings here: https://github.com/Ultimaker/Cura/wiki/Cura-Preferences-and-Settings-Locations. Depending on platform you can remove the 3.1 subfolders in these locations to have a clean 3.0.x install that will be used for the upgrade when starting 3.1 for the first time.

Totally can image you want to use 3.1 as soon as it comes out, so hopefully we can help you get back on track!

Thisismydigitalself, 2017-12-10:
removed folders from AppData. still no good.
uninstalled all versions. installed 3.0.4. ran OK. then tried installing 3.1 - did not run at all. splash screen of 3.1 until "setting up scenes" than nothing. 
ended up deleting all Ultimaker folders from Program files, installing 3.1 again. same - splash screen of 3.1 until "setting up scenes" than nothing.

ChrisTerBeke, 2017-12-10:
@LipuFei related to your fix from Friday?

@Thisismydigitalself could you post the log file here so we see the underlying issue?

Thisismydigitalself, 2017-12-10:
350Mb in size - how come log file includes few days back when i deleted all files and started from scratch?
deleted old logs. attached is from this last hour or so.

[CuraLog3.1-not Running.zip](https://github.com/Ultimaker/Cura/files/1546170/CuraLog3.1-not.Running.zip)

ChrisTerBeke, 2017-12-10:
The uploaded file is only 171kb in size.

Anyways, I'm seeing `ValueError: max() arg is an empty sequence`, which is something we've seen before and are in the process of fixing. If more people have it I'm sure we'll be releasing a patch version very soon.

ChrisTerBeke, 2017-12-10:
Related to https://github.com/Ultimaker/Cura/issues/2798?

ChrisTerBeke, 2017-12-10:
I'm closing this issue in favor of the already existing https://github.com/Ultimaker/Cura/issues/2798. Please keep an eye on that one to track the status of fixing this.

closed 2017-12-10 by ChrisTerBeke

Thisismydigitalself, 2017-12-10:
Thank U. Just reinstalled 3.0.4 which runs OK.