Ultimaker/Cura #2954: 3.1.0: Failing test suite

hroncok, 2017-12-10:
This is Fedora rawhide (to be 28), Python 3.6.3.

```console
$ python3 -m pip freeze
numpy==1.13.3
py==1.5.2
pytest==3.2.3
scipy==1.0.0
```

The following tests fail:

```
tests/Settings/TestCuraContainerRegistry.py::test_addContainerExtruderStack FAILED
tests/Settings/TestCuraContainerRegistry.py::test_addContainerGlobalStack FAILED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[ExtruderLegacy.stack.cfg-ExtruderStack] FAILED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[MachineLegacy.stack.cfg-GlobalStack] FAILED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[Left.extruder.cfg-ExtruderStack] FAILED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[Global.global.cfg-GlobalStack] FAILED
tests/Settings/TestCuraContainerRegistry.py::test_loadTypes[Global.stack.cfg-GlobalStack] FAILED
tests/Settings/TestCuraContainerRegistry.py::test_loadLegacyFileRenamed FAILED
tests/Settings/TestExtruderStack.py::test_addContainer ERROR
tests/Settings/TestExtruderStack.py::test_constrainUserChangesInvalid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainUserChangesInvalid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainUserChangesInvalid[container2] ERROR
tests/Settings/TestExtruderStack.py::test_constrainUserChangesValid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainUserChangesValid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesInvalid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesInvalid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesInvalid[container2] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesValid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityChangesValid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityInvalid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityInvalid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityInvalid[container2] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityValid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainQualityValid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainMaterialInvalid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainMaterialInvalid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainMaterialInvalid[container2] ERROR
tests/Settings/TestExtruderStack.py::test_constrainMaterialValid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainMaterialValid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainVariantValid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainVariantValid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainVariantInvalid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainVariantInvalid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_constrainDefinitionValid[container0] ERROR
tests/Settings/TestExtruderStack.py::test_constrainDefinitionValid[container1] ERROR
tests/Settings/TestExtruderStack.py::test_deserializeRemovesWrongInstanceContainer ERROR
tests/Settings/TestExtruderStack.py::test_deserializeRemovesWrongContainerClass ERROR
tests/Settings/TestExtruderStack.py::test_deserializeWrongDefinitionClass ERROR
tests/Settings/TestExtruderStack.py::test_deserializeMoveInstanceContainer ERROR
tests/Settings/TestExtruderStack.py::test_getPropertyFallThrough ERROR
tests/Settings/TestExtruderStack.py::test_insertContainer ERROR
tests/Settings/TestExtruderStack.py::test_removeContainer ERROR
tests/Settings/TestExtruderStack.py::test_setDefinitionByIdExists ERROR
tests/Settings/TestExtruderStack.py::test_setDefinitionByIdDoesntExist ERROR
tests/Settings/TestExtruderStack.py::test_setMaterialByIdExists ERROR
tests/Settings/TestExtruderStack.py::test_setMaterialByIdDoesntExist ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyUser[layer_height-value-0.1337] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyUser[foo-value-100] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyUser[support_enabled-value-True] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyUser[layer_height-default_value-0.1337] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyUser[layer_height-is_bright_pink-of course] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyOtherContainers[user-userChanges] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyOtherContainers[quality_changes-qualityChanges] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyOtherContainers[quality-quality] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyOtherContainers[material-material] ERROR
tests/Settings/TestExtruderStack.py::test_setPropertyOtherContainers[variant-variant] ERROR
tests/Settings/TestExtruderStack.py::test_setQualityByIdExists ERROR
[... 68 lines cut ...]
tests/Settings/TestGlobalStack.py::test_setPropertyUser[layer_height-value-0.1337] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyUser[foo-value-100] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyUser[support_enabled-value-True] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyUser[layer_height-default_value-0.1337] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyUser[layer_height-is_bright_pink-of course] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyOtherContainers[user-userChanges] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyOtherContainers[quality_changes-qualityChanges] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyOtherContainers[quality-quality] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyOtherContainers[material-material] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyOtherContainers[variant-variant] ERROR
tests/Settings/TestGlobalStack.py::test_setPropertyOtherContainers[definition_changes-definitionChanges] ERROR
tests/Settings/TestGlobalStack.py::test_setQualityByIdExists ERROR
tests/Settings/TestGlobalStack.py::test_setQualityByIdDoesntExist ERROR
tests/Settings/TestGlobalStack.py::test_setQualityChangesByIdExists ERROR
tests/Settings/TestGlobalStack.py::test_setQualityChangesByIdDoesntExist ERROR
tests/Settings/TestGlobalStack.py::test_setVariantByIdExists ERROR
tests/Settings/TestGlobalStack.py::test_setVariantByIdDoesntExist ERROR
tests/Settings/TestGlobalStack.py::test_smoke_findDefaultVariant ERROR
tests/Settings/TestGlobalStack.py::test_smoke_findDefaultMaterial ERROR
tests/Settings/TestGlobalStack.py::test_smoke_findDefaultQuality ERROR
```

139 is `cura/Settings/CuraContainerStack.py:48: IndexError`

full log at https://kojipkgs.fedoraproject.org//work/tasks/5709/23625709/build.log

ChrisTerBeke, 2017-12-10:
We're already working on fixing this (keep an eye out for master branch).

hroncok, 2017-12-10:
Thanks @ChrisTerBeke. As said in similar [Uranium PR](https://github.com/Ultimaker/Uranium/pull/320#issuecomment-350546338), could you please develop a mechanism to prevent releasing new versions with failing tests? This is happening every release. I can help, if you want.

ChrisTerBeke, 2017-12-10:
It's always a race between releasing on the deadline and delivering the best code possible. Sometimes one ones, sometimes the other. We try to keep the tests working at all times, but when we don't help is always appreciated!

hroncok, 2017-12-10:
@ChrisTerBeke Could you please point me to the fix?

> We try to keep the tests working at all times, but when we don't help is always appreciated!

Well what I meant is helping you to set up a process that would prevent this in the future. Would you be interested in a PR that adds Travis CI?

hroncok, 2017-12-10:
@ChrisTerBeke this is still not fixed on master.

ChrisTerBeke, 2017-12-10:
We have an internal CI system in place for Cura, which also tests Uranium as they're very connected when running as application. Not sure if others in the team agree with a public Travis connection. CC @Ghostkeeper @ArjanSmit

hroncok, 2017-12-10:
But your internal CI does not publish build status to GtiHub. Nor does it prevent you to merge PR with broken tests, or from creating yet another release with broken tests. It's not about the tool (Travis or Jenkins, or whatnot), it's about the process. And having the CI setup to prevent this kind of situation from happening again might help.

hroncok, 2017-12-10:
I mean, you could say: "You know @hroncok, we don't actually care whether the tests are broken or not, just don't run them when you package Cura for Fedora and stop reporting the failures." You have all right to say so. But I just don't see the point of having a testsuite, if it's not taken care for.

This is as simple as creating a rule: **A push to master that breaks the tests if forbidden** (of course, there might be situations where this rule is not followed, but an explanation shall be provided in that case). Whether this is checked by a person or by a robot is an implementation detail.

ChrisTerBeke, 2017-12-10:
I'm not the one working on the process, I'm sure there are good reasons for our current setup. I'll throw it in the group and maybe the people tagged above will respond, they are a bit longer and more involved in the process. I'm sure we do want working tests as we are still writing them for larger new features.

hroncok, 2017-12-10:
OK, thank you.

BTW could you please reopen this issue? It's still relevant.

alekseisasin, 2017-12-13:
So far as I know the last test which failed should be fixed for now, do you ( @hroncok  ) still have fails in test ?

hroncok, 2017-12-13:
Both master and 3.1 branch still fails. Could you please point me to the specific commit that should have fixed the failures?

alekseisasin, 2017-12-13:
this was not related to one commit, it was a big merge with around 300 commits which affected whole application, CURA-4243. I will retest them again, if there is still issue then, yes it should be solved.

Ghostkeeper, 2017-12-30:
The unit tests are working at the moment:

    cherubim@Cherub:~/Projects/Cura/build$ PYTHONPATH=/home/cherubim/Projects/Uranium:/home/cherubim/dev/inst/lib/python3/dist-packages ctest --verbose
    UpdateCTestConfiguration  from :/home/cherubim/Projects/Cura/build/DartConfiguration.tcl
    UpdateCTestConfiguration  from :/home/cherubim/Projects/Cura/build/DartConfiguration.tcl
    Test project /home/cherubim/Projects/Cura/build
    Constructing a list of tests
    Done constructing a list of tests
    Checking test dependency graph...
    Checking test dependency graph end
    test 1
        Start 1: pytest-main

    1: Test command: /home/cherubim/dev/inst/bin/python3.5 "-m" "pytest" "--junitxml=/home/cherubim/Projects/Cura/build/junit-pytest-main.xml" "/home/cherubim/Projects/Cura/tests"
    1: Environment variables: 
    1:  PYTHONPATH=/home/cherubim/Projects/Cura:/home/cherubim/Projects/Cura/../Uranium:/home/cherubim/Projects/Uranium:/home/cherubim/dev/inst/lib/python3/dist-packages
    1: Test timeout computed to be: 9.99988e+06
    1: ============================= test session starts ==============================
    1: platform linux -- Python 3.5.2, pytest-3.2.2, py-1.4.34, pluggy-0.4.0
    1: rootdir: /home/cherubim/Projects/Cura, inifile: pytest.ini
    1: collected 164 items
    1: 
    1: ../tests/TestArrange.py .............
    1: ../tests/TestMachineAction.py .
    1: ../tests/TestProfileRequirements.py ....
    1: ../tests/Settings/TestCuraContainerRegistry.py .....
    1: ../tests/Settings/TestExtruderStack.py ...........................s....s.......................
    1: ../tests/Settings/TestGlobalStack.py ..........................................s....s.....................................
    1: 
    1: - generated xml file: /home/cherubim/Projects/Cura/build/junit-pytest-main.xml -
    1: ==================== 160 passed, 4 skipped in 1.16 seconds =====================
    1/4 Test #1: pytest-main ......................   Passed    1.49 sec
    test 2
        Start 2: pytest-VersionUpgrade26to27

    2: Test command: /home/cherubim/dev/inst/bin/python3.5 "-m" "pytest" "--junitxml=/home/cherubim/Projects/Cura/build/junit-pytest-VersionUpgrade26to27.xml" "/home/cherubim/Projects/Cura/plugins/VersionUpgrade/VersionUpgrade26to27"
    2: Environment variables: 
    2:  PYTHONPATH=/home/cherubim/Projects/Cura/plugins/VersionUpgrade/VersionUpgrade26to27:/home/cherubim/Projects/Cura:/home/cherubim/Projects/Cura/../Uranium:/home/cherubim/Projects/Uranium:/home/cherubim/dev/inst/lib/python3/dist-packages
    2: Test timeout computed to be: 9.99988e+06
    2: ============================= test session starts ==============================
    2: platform linux -- Python 3.5.2, pytest-3.2.2, py-1.4.34, pluggy-0.4.0
    2: rootdir: /home/cherubim/Projects/Cura, inifile: pytest.ini
    2: collected 13 items
    2: 
    2: ../plugins/VersionUpgrade/VersionUpgrade26to27/tests/TestVersionUpgrade26to27.py .............
    2: 
    2:  generated xml file: /home/cherubim/Projects/Cura/build/junit-pytest-VersionUpgrade26to27.xml 
    2: ========================== 13 passed in 0.44 seconds ===========================
    2/4 Test #2: pytest-VersionUpgrade26to27 ......   Passed    0.75 sec
    test 3
        Start 3: pytest-VersionUpgrade27to30

    3: Test command: /home/cherubim/dev/inst/bin/python3.5 "-m" "pytest" "--junitxml=/home/cherubim/Projects/Cura/build/junit-pytest-VersionUpgrade27to30.xml" "/home/cherubim/Projects/Cura/plugins/VersionUpgrade/VersionUpgrade27to30"
    3: Environment variables: 
    3:  PYTHONPATH=/home/cherubim/Projects/Cura/plugins/VersionUpgrade/VersionUpgrade27to30:/home/cherubim/Projects/Cura:/home/cherubim/Projects/Cura/../Uranium:/home/cherubim/Projects/Uranium:/home/cherubim/dev/inst/lib/python3/dist-packages
    3: Test timeout computed to be: 9.99988e+06
    3: ============================= test session starts ==============================
    3: platform linux -- Python 3.5.2, pytest-3.2.2, py-1.4.34, pluggy-0.4.0
    3: rootdir: /home/cherubim/Projects/Cura, inifile: pytest.ini
    3: collected 13 items
    3: 
    3: ../plugins/VersionUpgrade/VersionUpgrade27to30/tests/TestVersionUpgrade27to30.py .............
    3: 
    3:  generated xml file: /home/cherubim/Projects/Cura/build/junit-pytest-VersionUpgrade27to30.xml 
    3: ========================== 13 passed in 0.05 seconds ===========================
    3/4 Test #3: pytest-VersionUpgrade27to30 ......   Passed    0.31 sec
    test 4
        Start 4: pytest-VersionUpgrade25to26

    4: Test command: /home/cherubim/dev/inst/bin/python3.5 "-m" "pytest" "--junitxml=/home/cherubim/Projects/Cura/build/junit-pytest-VersionUpgrade25to26.xml" "/home/cherubim/Projects/Cura/plugins/VersionUpgrade/VersionUpgrade25to26"
    4: Environment variables: 
    4:  PYTHONPATH=/home/cherubim/Projects/Cura/plugins/VersionUpgrade/VersionUpgrade25to26:/home/cherubim/Projects/Cura:/home/cherubim/Projects/Cura/../Uranium:/home/cherubim/Projects/Uranium:/home/cherubim/dev/inst/lib/python3/dist-packages
    4: Test timeout computed to be: 9.99988e+06
    4: ============================= test session starts ==============================
    4: platform linux -- Python 3.5.2, pytest-3.2.2, py-1.4.34, pluggy-0.4.0
    4: rootdir: /home/cherubim/Projects/Cura, inifile: pytest.ini
    4: collected 18 items
    4: 
    4: ../plugins/VersionUpgrade/VersionUpgrade25to26/tests/TestVersionUpgrade25to26.py ..................
    4: 
    4:  generated xml file: /home/cherubim/Projects/Cura/build/junit-pytest-VersionUpgrade25to26.xml 
    4: ========================== 18 passed in 0.39 seconds ===========================
    4/4 Test #4: pytest-VersionUpgrade25to26 ......   Passed    0.71 sec

    100% tests passed, 0 tests failed out of 4

    Total Test time (real) =   3.27 sec

ChrisTerBeke, 2017-12-30:
Closing as the unit tests were repaired. We'll keep a closer eye on not breaking them again.

closed 2017-12-30 by ChrisTerBeke