Ultimaker/Cura #2955: Cura 3.1.0 not showing "Material View"

claude1000, 2017-12-10:
Cura 3.1.0 not showing "Material View"

fieldOfView, 2017-12-10:
Needs more info. What printer are you using, what do you call "Material View"?

claude1000, 2017-12-10:
Ultimaker 3  

[image]

Cura 3.1
- computer at home : can't see material color 
- computer at my job : I see  material color  (see attached picture)

ChrisTerBeke, 2017-12-10:
Are they both the same OS and version? Do they have the same Cura machine configuration?

Ghostkeeper, 2017-12-30:
I can't reproduce this. "Material Color" is an option in the "Color scheme" drop-down and when selected it properly displays the colours of the current materials.

ChrisTerBeke, 2018-05-23:
No new info added, cannot reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke