Ultimaker/Cura #2956: how to resolved the unwanted lines on surface

claude1000, 2017-12-10:
how to resolved the unwanted lines on surface

https://ultimaker.com/en/community/52344-unwanted-lines-on-surface

smartavionics, 2017-12-10:
Did you try those settings I gave you on the forum? Perhaps they will provide a solution to your problem?

claude1000, 2017-12-10:
I launch to day a print with your setup tomorrow I will return back to see 
I will inform you if it's resolved
rgs
Claude

smartavionics, 2017-12-11:
Maybe related to https://github.com/Ultimaker/CuraEngine/issues/625

Ghostkeeper, 2017-12-30:
I think I remember that we won't z-hop or retract for very small travel moves, smaller than 2 line widths or so. Maybe that could be what's happening here.

ianpaschal, 2018-05-28:
Didn't hear back from OP for 6 months so closing.

closed 2018-05-28 by ianpaschal