Ultimaker/Cura #2957: baud rate detection failed Cura 3.1 Marlin 1.1.6 Creality CR-10S

lastsnare, 2017-12-10:
discovered that Cura reports "baud rate detection failed" after I upgraded my Creality CR-10S and CR-10 S5 to TH3D Marlin 1.1.6.
I noticed that I could connect at 115200 baud with S3D and with Printrun, but Cura would fail.
So, after some tinkering, I changed time.sleep value to 10.0 seconds (I just picked a number higher than the default of 1.5 seconds, to prevent "talking to bootloader").
after increaseing the timeout value I was able to connect again with Cura.
In Windows, this is what I edited:
C:|Program Files\Ultimaker Cura 3.1\plugins\USBPrinting and the file is USBPrinterOutputDevice.py
Using Notepad++ I was able to identify the line numbers in the Cura.log file while troubleshooting.
by changing line 340 "time.sleep" value from "1.5" to "10.0", I was able to get Cura to connect to my printer again.
It could be that the Marlin 1.1.6 hex that TH3D assembled takes a little too long to boot and hits the timeout before Cura gives up.  But, it did not seem to matter if I left it connected and sitting for a while once the printer had fully booted it's firmware, it did not appear to retry and eventually connect.  modifying the timeout value seems to have fixed the issue in this specific case.

Ghostkeeper, 2017-12-30:
Changing that sleep time is not really an option for us though, because depending on the baud rate of the printer it could take up to 7 * 8.5s = 59.5s longer to connect to the printer. We wouldn't want people to have to wait a minute extra before their USB connection finally gets picked up by Cura.

Maybe we'd need to make this variable, per printer? Or perhaps even better, we'd need to come up with a more clever time-out scaling technique that first tries small time-outs and gradually increases the time-out between tries. I wonder why that sleep is even inside the loop over baud rates though...

fieldOfView, 2017-12-30:
I think what legacy Cura did was to use a short-ish timeout when auto-detecting, but a longer timeout when a baudrate was specified. IE: try harder when the user says it is so.

I will give the whole "manual port and baudrate" thing a new go once the network printing branch has stabilized. See https://github.com/Ultimaker/Cura/pull/1688, which was an initial stab at implementing https://github.com/Ultimaker/Cura/issues/1049

closed 2018-03-02 by ChrisTerBeke