Ultimaker/Cura #2958: Missing detail

NovaDenied, 2017-12-10:
Some detail in solid view is not showing up in layer mode and thus is not being printed on the model. The detail is there in cura 2.5 but not 3.0 or 3.1. Some detail thats there is thinner than on cura 2.5 as well.
[image]
[image]

NovaDenied, 2017-12-10:
The detail is in cura 2.5.
[image]

Vandrasc, 2017-12-11:
Hi @NovaDenied, is the Print Thin Walls setting enabled? You can search for it while in the Custom tab in the search box. Also, it is possible to attach your model so we can have a look? 
Thank you!

NovaDenied, 2017-12-11:
No it wasn't but it seems to be working now I dont understand what happened but I do know that cura crashed and then reset my settings and it worked since then. Must of been a glitch.

ChrisTerBeke, 2017-12-11:
Maybe related to our 3.1 beta settings bug. After cleaning the settings folder it should then be solved indeed.

closed 2017-12-11 by ChrisTerBeke

NovaDenied, 2017-12-11:
how do i do that?

Ghostkeeper, 2017-12-30:
What he meant is to basically reset Cura to factory defaults, removing all profiles and custom settings and such. You can do that by opening Cura and going to Help -> Show Configuration Folder. Then close Cura. Then delete the parent of that folder it brings you to (called `cura`).

Or less intrusive, delete only the changes that were made while using Cura 3.1 by deleting just the `3.1` folder. The glitch you saw could be caused by using the 3.1 beta that left the configuration in a broken state.