Ultimaker/Cura #2959: hole

carleas, 2017-12-10:
hello, 

i have a problem when i'm go to print a figure with one central hole.
In the model the figure have this, but when i print or i put the layer view the printer print all and the hole disappears.

What its the possible problem?

Tanks.

Vandrasc, 2017-12-11:
Hi @carleas, could you attach your file/model here so we can give it a try? I loaded different models but I did not encounter this issue.
As a workaround, you could set your Top Layers to 0 ( CustomTab/Shell/TopLayers )
Thank you!

carleas, 2017-12-13:
Hi! I send a E-mail with de model.

I’m waiting for the Solution.

Tanks

nallath, 2017-12-19:
This is probably because your model is not manyfold. Check the model in x-ray. If there are red parts visible, the model has issues.

carleas, 2017-12-21:
Hi, tanks for your answer,

I draw another model and i check in x-Ray, the problem it's the same.
I have in 2 diferent dimencions and the problem persist.

Plis have me a solution. I wanted to do it for a chrismas gift.

[image: Imatge inserida 2] [image: Imatge inserida 3][image: Imatge
inserida 5]

Ghostkeeper, 2018-01-15:
You said you sent an e-mail with the model. Where did you send it to? Maybe you could put it in a .zip and upload it here on Github?

Ghostkeeper, 2018-01-15:
You could try fixing the model in MeshLab.

closed 2018-03-02 by ChrisTerBeke