Ultimaker/Cura #2961: Cura 3.1.0 Repetative Unresponsiveness Still Not Resolved

dkingsjr, 2017-12-11:
I am running Cura 3.1.0 on a Windows 10 PC. Cura is continuing to become unresponsive. Unresponsiveness happens often. about every 30 seconds, sometimes a little longer. Program initialization is also still slow as well.

ChrisTerBeke, 2017-12-11:
Can you give us the logs and specs of your computer? We can't reproduce it so far.

lucdesign, 2017-12-13:
cura 3.1 is almost unusably unresponsive on my mac. all versions after 1 (15.xx.xx) have the same problem. in V 3.1, clicking on "prepare" does not seem to do anything most of the time. sometimes, without any reason, it then slices. changing a setting takes forever to register, 10 seconds are not rare. the prefs window jumps in front of the printer settings when the program is de- and reactivated.

ChrisTerBeke, 2017-12-13:
Can you post some system specs or logs? Without this we can't do anything to reproduce and maybe fix the problem.

regmigrant, 2017-12-17:
I  have the same issue and thought it was because I'm using Windows fast ring version but as it's happening to others:
Win 10, enterprise (build 17046), 64 bit, Core i7 920 @ 2.6gHz, 12Gb, Samsung 500gb SSD in Mirror Raid (on board support). Further detail in attachment:
[software.txt](https://github.com/Ultimaker/Cura/files/1565442/software.txt)
[sysinfo.txt](https://github.com/Ultimaker/Cura/files/1565443/sysinfo.txt)

I have Hypervisor enabled but Cura is not in a virtualised environment

I tried disabling malware monitoring and unchecking the automatic slice option but I'm getting the same brown outs as the others.

Ghostkeeper, 2018-01-04:
I've seen this when lots of models are loaded on one build plate (>20). But with any normal print of 4 or less it works quite fine even on my sluggish old laptops.

If it happens periodically that could be an indicator that writing to the file system is slow, because settings are periodically auto-saved (as long as the user interacts with the program at least).

ChrisTerBeke, 2018-05-23:
No new info, cannot reproduce. Closing.

closed 2018-05-23 by ChrisTerBeke