Ultimaker/Cura #2962: Request For Next Update: Eliminate The "Stage Light" Effect.

dkingsjr, 2017-12-11:
One thing I've noticed, is that there is a horrible light reflection effect going on in Cura. It is WORSE with the white theme. and gets a little better with the dark theme...however, i think it should be eliminated all together. Or at least we should have the option to turn it on and off as we see fit. I am not entirely sure why there is a need for a light reflection off the model loaded into Cura, but it makes seeing certain details horribly difficult to see, and is a little bit taking on the eyes.

alekseisasin, 2017-12-12:
Maybe can You please send a print screen (highlight areas) of the area which in your opinion should be customized ?

fieldOfView, 2017-12-12:
I think the request is to remove (or tone down) the specular highlight

Ghostkeeper, 2017-12-30:
Sometimes it's just necessary to have a specular reflection in order to make out details on the surface. For instance, here is a wavy sort of surface without specular reflections ([this model](https://www.thingiverse.com/thing:2139138)):
[image]

So I am of the opinion that the specular reflection is useful and important even.

The white theme has the same specular reflection as the dark theme, so that could be placebo or only perceptual.

ianpaschal, 2018-03-09:
Since this is not hard for the user to edit themselves and not really an issue with the software, I'm closing this.

closed 2018-03-09 by ianpaschal

FrozenFold, 2020-02-03:
It would still be nice to have a setting within the program for people like me who don't know how they are expected to change it once in the file folder for the shaders.