Ultimaker/Cura #2963: Nozzle offset cannot be set correctly

PatSmith1969, 2017-12-11:
I have a chimera print head. I need to set nozzle 2 exactly 18mm off-set on the X axis. When I do so, it changes BOTH nozzles to 18mm off-set in the printer properties page. Setting Nozzle 1 back to 0 offset, changes BOTH nozzles to 0 off-set.

ChrisTerBeke, 2017-12-11:
Hi. Did you use the 3.1 beta version before using the stable? If so, try removing the 3.1 user data folder and then start the 3.1 stable. If not, please let us know so we can investigate!

an-tech, 2017-12-11:
Same problem. I've used 3.1 beta. Tried delete data in users/userfolder/appdata/local/cura3.1 and reinstall with no success

an-tech, 2017-12-11:
Workaround: You can set extruder2 as active. Then you can change values.

AlfredVischermann, 2017-12-17:
I've got the same problem using Cure 3.1, 
setting extruder 2 as the active one didn't work for me

an-tech, 2017-12-17:
Exactly. It's saving values but it appears in both tabs ext1 and ext2. When you change to extruder1 it shows ext1 values also in both tabs. Not a big problem but watch out for another bug when using ext2 supports #2976.

ChrisTerBeke, 2017-12-19:
This will be fixed in the next release (3.2), of which the beta is planned in January.

closed 2017-12-19 by ChrisTerBeke