Ultimaker/Cura #2966: Cura 3.1 doesn't recognize Sain Smart Anet A8 on Mac 10.13.1

DavidMnDenver, 2017-12-11:
I have a Mac desktop running OS 10.13.1 and I just purchased a Sain Smart Anet A8 connected trough my USB port. I've tried different cables, ports (I even tried a direct port on the back of the Mac), everything I could think of, but neither Cura nor the Mac seem to recognize that the 3D printer is even there. It doesn't show in my USB tree or the printer list.

I tried to find some kind of driver online but after a week of searching came up with only a single (and, therefore, suspect) entry which claims that they don't know if their driver will work and, as it's from Eastern Europe, I'm hesitating to download a possible virus or malware without a realistic chance of a solution.

Sain Smart has shrugged their electronic shoulders and said to use a SD card instead. Yes, I know I can do that but I want the more flexible control of operating from a computer.

Any ideas?

ChrisTerBeke, 2017-12-11:
This sounds more of a driver/firmware problem than a Cura problem. Usually we don't give support for these situations.

I have an anet myself an installed https://blog.sengotta.net/signed-mac-os-driver-for-winchiphead-ch340-serial-bridge/ on my Mac, then restarted. After that it was picked up correctly.

closed 2017-12-11 by ChrisTerBeke

DavidMnDenver, 2017-12-11:
Thanks. You’re aces. Works perfectly.