Ultimaker/Cura #2967: Slicing simple pyramid top model causes a lot of travels

bipsendk, 2017-12-11:
When slicing the attached model, a lot of travelling between printing is taking places on each layer - instead of just moving in circles towards the top.

Is there a way to optimize the slicing of this model ?

[Hoved-2.zip](https://github.com/Ultimaker/Cura/files/1549081/Hoved-2.zip)

smartavionics, 2017-12-11:
Most of those travels are caused by the filling between the walls. Cura doesn't really make a good job of that so, personally, I would disable the filling (set fill gaps between walls to nowhere). The travels can be reduced further by setting a user specified z-seam position so that the seams on both the inner and outer cones are aligned. Finally, do enable optimize wall printing order as the optimizer does try to avoid travels where it can. Hope this helps.

smartavionics, 2017-12-11:
[image]

closed 2017-12-30 by Ghostkeeper