Ultimaker/Cura #2968: Cura 3.1 crash when switching to .1 layer height

tdub415, 2017-12-12:
When trying to select the .1 layer height, I get a blank window that pops up, and upon clicking the 'x' button on it, cura completely closes:
[image]

After this happens, I am unable to even open cura.  As soon as I open it, the window pops back up and the process repeats.

Attached is the log file after this loop has started.
[cura.log](https://github.com/Ultimaker/Cura/files/1550750/cura.log)

tdub415, 2017-12-12:
This is a fresh install of Cura 3.1, I am using the profiles provided by Monoprice for the Maker Select Mini V2, adding to my %appdata% folder. at cura/3.1/definitions if that helps with diagnosing

profiles found at the bottom of the page: https://www.monoprice.com/product?p_id=21711

[image]

tdub415, 2017-12-12:
I have also noticed that if I delete my machine_instance, i can break the crash loop, and upon examining the profiles, it appears that the "Fine" profile, which looks like it should be the 0.1 layer height profile is completely blank, while the other profiles at least have basic information in them.

ChrisTerBeke, 2017-12-12:
ping @LipuFei

LipuFei, 2017-12-12:
Hi @tdub415 , seems to be a problem with the `monoprice_select_mini.def.json` file. You can overwrite your current file with this one [monoprice_select_mini.def.json.zip](https://github.com/Ultimaker/Cura/files/1551511/monoprice_select_mini.def.json.zip), or fix it manually with a text editor:

 - Change `"default_value": "0.10"` to `"default_value": 0.10` (remove the double quotes around the number) on line 40 for `"layer_height"`.

Also remove the directory `AppData\Local\cura` before you start cura. Please note that it's in `Local`, **NOT** `Roaming`.

LipuFei, 2017-12-12:
@tdub415 I have also contacted Monoprice so they can fix the problem too.

tdub415, 2017-12-12:
Awesome, that fixed it!  Thank you so much.

closed 2017-12-12 by LipuFei