Ultimaker/Cura #2971: [3.1] S{material_print_temperature_layer_0} gives incorrect data

MaxRink, 2017-12-13:
As you are able to see here https://i.imgur.com/fE8xK22.jpg the variable for the first layer temp gives an incorrect result. THe Temp for the first layer is set to 230°C, but the value in the GCODE is 210. This V alue is entered nowhere, neither in the specific filament settings nor elsewhere for this print

ChrisTerBeke, 2017-12-13:
Duplicate of https://github.com/Ultimaker/Cura/issues/1296.

closed 2017-12-13 by ChrisTerBeke

fieldOfView, 2017-12-13:
Yeah, no, not really a duplicate. It may be the same underlying issue, but Cura 3.1 broke all start gcode using {material_print_temperature*} for single extrusion printers.

As a workaround, a user can remove the lines with {material_print_temperature*} and CuraEngine will insert its own heatup sequence before the start gcode, but that is not an option for people who need to eg turn on their PSU before heating up.

ChrisTerBeke, 2017-12-13:
It's the same underlying issue, it's now just more visible as single extrusion machines also have an extruder stack. It's definitely back on the urgent to fix list as well. To keep all discussion about the issue in one place, I marked this as duplicate.

alexwhittemore, 2018-01-30:
Well, at least I'm not crazy.