Ultimaker/Cura #2972: Question - move the wall order optimization setting out of experimental?

smartavionics, 2017-12-13:
As I haven't yet seen any issues posted related to the wall print order optimization doing bad things and I think that it is getting used, is it reasonable to move it out of the experimental section into the shell section? And if so, do we want to keep it disabled by default or promote it to be enabled by default?

ChrisRiddell, 2017-12-13:
I've been using it and have not seen any adverse effects, Only improvements in the wall print order.

I can not really see any reason it should not be enabled by default, the only time you would really need to turn it off is for edge cases where it increases print time?

Ghostkeeper, 2018-01-01:
I've been trying to convince our Material & Processing guys to allow us to do this but they don't have time to test with it yet. Discussing with our colleague we can move it out of experimental but enabling it by default really requires their consent, or we'd be changing the working of the engine considerably while they are working on their profiles and they'd have to redo all that.

I don't really like to have this as a setting. It should just be always optimised, always grouping adjacent walls together.

smartavionics, 2018-01-01:
Once it left beta, I don't think I have seen a single issue that is due to the optimisation code so either people aren't using it or it is working OK. I understand the position of your colleagues so I would be happy to settle for moving out of experimental for now. I shall create a little PR to move it.

fieldOfView, 2018-01-01:
SliceInfo data should tell us if people are using it.

smartavionics, 2018-01-01:
Does that data actually include the settings that people are using to slice?

nallath, 2018-01-03:
A subsection of the settings. So the wall order optimisation isn't in there.

smartavionics, 2018-01-03:
> A subsection of the settings. So the wall order optimisation isn't in there.

Sorry, too early for me, what do you mean by that?

nallath, 2018-01-03:
That we don't send all settings, we just send a few of them. We don't send any settings regarding the wall order optimisation. This means we don't actually know if people use it.

smartavionics, 2018-01-03:
Ah, right. Well that's a shame.

Ghostkeeper, 2018-01-18:
Here's a list of stuff that gets sent via the anonymous slice statistics: https://github.com/Ultimaker/Cura/blob/dd8d21fd341cffb16552148f3511cf38accaa516/plugins/SliceInfoPlugin/SliceInfo.py#L72

smartavionics, 2018-01-18:
It's now been move, thanks, closing...

closed 2018-01-18 by smartavionics