Ultimaker/Cura #2973: Coasting not always replace the end of extrusion path with travel.

ChrisRiddell, 2017-12-13:
Cura 3.1
Windows 10
Ultimaker2+

Something i noticed while placing with coasting with filament that's a little oozy was that coasting is not always replacing the end of an extrusion path with a travel path.

[image]
1 you can see it's not doing coasting on the outer outline but is for 2 like I'd expect.

[image]
and this one you can see its not doing coasting at all before it moves off.

Ghostkeeper, 2018-01-01:
I'm not able to reproduce this. All paths, the long ones and the short ones, seem to have gaps in there that get replaced by travel moves. Only when a line is very short, the Minimum Volume Before Coasting comes into play and prevents the coasting on that perimeter. But in your screenshot it looks like it's the longer path that gets no coasting rather than the shorter path.

Maybe it's specific to certain models? Could you share a model that reproduces this?

Looking closely at your second screenshot, it looks almost like the wall is broken into pieces over there. There seems to be two seams: One where the original "1" is pointing in the first screenshot, and one a little bit lower where you seen the travel moves in the second screenshot. Maybe then the Minimum Volume Before Coasting kicked in and prevented the coasting in that perimeter?

ChrisRiddell, 2018-02-05:
Had anther play it seems to be related to travel movements that don't have a retraction (something to do with combing maybe?)

closed 2018-02-28 by ChrisRiddell