Ultimaker/Cura #2975: CURA 3.1.0 - Skirt bug

thercman, 2017-12-13:
The skirt only surrounds support area while using supports. This creates unwanted filament under the model. Without supports it functions as normal.

nallath, 2017-12-19:
We're going to need a bit more to figure this out. Could you show some screenshots of what you mean? Could you also share some info on your setup (eg; what machine, what settings, etc)

CraigMason, 2017-12-30:
I have the same problem.

Simply ticking support will generate a different skirt layout, which goes underneath layer 1.

[image]

[image]

Ghostkeeper, 2018-01-17:
The cause of this is usually that the model is not entirely flat on the build plate. There are two causes for the model not being flat on the build plate:
* Sometimes there is a very thin bit on the bottom side (like noise of a 3D scan or a thin ridge) that is too thin to print. The model is positioned on the build plate but it's not flat so when actually slicing the model floats in air.
* There was a rounding error in the model dimensions that we introduced to increase processing speed when manipulating objects in the build volume. This rounding error has been practically removed for 3.2 so that is considered fixed now.

When the model is not flat on the build plate for either of these reasons, after slicing, Cura will remove all layers on the bottom that are completely empty so that the model is not printed in mid-air (precisely for the aforementioned reasons). However when you add support, then the model is still in mid-air when adding support and the support goes all the way to the build plate, so you'll have support on the first layer. There are no empty first layers then so nothing gets removed. In your case you'll then have one single layer at the bottom with just some support, and it looks like the skirt goes underneath the model because it actually goes on the layer below the model.

A workaround if your model is incorrect is to move the object down by a fraction of a millimetre.

If you're convinced that the model is actually flat and there are no rounding errors involved, we're going to need a model file to reproduce it with, because this doesn't happen with the models that I've tried.

CraigMason, 2018-01-17:
[This](https://www.thingiverse.com/thing:1657541) was the Thingiverse item I was printing in my screenshots above.

The file is `headphone_holder_short.stl`, rotated 90 degrees onto its side.

[Monster_Mouth_Headphone_Holder_-_Clampable.zip](https://github.com/Ultimaker/Cura/files/1638668/Monster_Mouth_Headphone_Holder_-_Clampable.zip)

CraigMason, 2018-01-17:
What seems to be clear is the skirt is only tracing the boundaries of the supports on my previous screenshots. Merely toggling the 'generate supports' option will switch between the skirt behaviours.

Ghostkeeper, 2018-01-17:
Looks like it's indeed fixed in 3.2:
[image]

closed 2018-01-17 by Ghostkeeper