Ultimaker/Cura #2977: Wall number makes diferent infill

ciberwizz, 2017-12-14:
Using the same settings only diferent wall thickness creates very diferent infill. Next Layer will print in mid-air.

[image]
[image]

Ghostkeeper, 2018-01-01:
That's not infill, that's skin. Because you reduced the number of walls, there is more of the inside being exposed to air at the top and it needs to create more skin. Or am I seeing this wrong? It's a bit hard to see, and I also don't see anything that would really print in mid-air.

ianpaschal, 2018-03-09:
Needs more info but never got it.

closed 2018-03-09 by ianpaschal