Ultimaker/Cura #2979: Master branch crash when updating a profile

smartavionics, 2017-12-14:
When I try to update a profile, I am getting this:
````
2017-12-14 10:17:24,361 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught error has occurred!
2017-12-14 10:17:24,363 - CRITICAL - cura.CrashHandler.__init__ [65]: Traceback (most recent call last):
2017-12-14 10:17:24,363 - CRITICAL - cura.CrashHandler.__init__ [65]:   File "/opt/cura/lib/python3/dist-packages/cura/Settings/ContainerManager.py", line 517, in updateQualityChanges
2017-12-14 10:17:24,365 - CRITICAL - cura.CrashHandler.__init__ [65]:     if not quality_changes or quality_changes.isReadOnly():
2017-12-14 10:17:24,365 - CRITICAL - cura.CrashHandler.__init__ [65]: AttributeError: 'InstanceContainer' object has no attribute 'isReadOnly'

````

smartavionics, 2017-12-14:
BTW, when the crash occurs an empty dialog pops up, not much use, is it?
[image]

Vandrasc, 2017-12-14:
Yes, it is a known issue, introduced by the printer definition. We will look into it, thank you for reporting it.

smartavionics, 2017-12-15:
Shit, now I can't even slice using the latest master branch code, I get this when I press the prepare button:
````
qml: Could not start or stop slicing TypeError: Cannot call method 'forceSlice' of undefined
````

I wish people would test new code for major breakage before committing to the master branch!

smartavionics, 2017-12-15:
I bet it's 47e5303a893dd5b1879a9ecd4f762278807961fd that's causing the breakage, what do you think @ChrisTerBeke ?

ChrisTerBeke, 2017-12-15:
I think that try-catch is fine, it's probably the way that backend is now connected via a property instead of a slot. I'll check it out. Reason I didn't find that is because I always use auto-slicing.

ChrisTerBeke, 2017-12-15:
Fixed, see latest commits in Cura and Uranium master branches.

smartavionics, 2017-12-15:
Thanks @ChrisTerBeke, that's fixed that little problem - perhaps now you could wave your magic wand over the original issue which is that when I try to update a profile it crashes (details are at top of this page).

ChrisTerBeke, 2017-12-15:
That's part of a refactor that some of us are working on (but not me, busy with other things). It's in the current sprint though so it'll get fixed soon.

smartavionics, 2017-12-15:
OK, well, it's a bit anti-social leaving the master branch broken. Why can't they commit it all in one session?

ChrisTerBeke, 2017-12-15:
They did, but there's always one or two bugs after merging something that large that slips through code review and component testing. Ofc it's not nice to leave master 'broken', but then again it's not meant to be stable for end-users either.

smartavionics, 2017-12-18:
This looks like it has been fixed now, thanks, closing...

closed 2017-12-18 by smartavionics

jackha, 2017-12-19:
Yeah I fixed both yesterday ^^. Thanks for reporting