Ultimaker/Cura #2982: [Feature request / Idea] Outer Wall Layer Height Setting

mpsdskd, 2017-12-15:
TL;DR: Separate setting for outer wall layer height, to speed up printing while still having hight resolution (similar to the "Infill Layer thickness" setting) 

All 3D-printer users want their print to finish as fast as possible, but this wish conflicts with the wish to have a high resolution (-> low layer height).
That's where my idea comes in: 
__How about a seperate layer height setting for the outer shell?__

As an example with the default cura settings I loaded a 20x20x20 cube, which takes 
52 min with "Fine - 0.1mm", 
outer wall 13 mins
(Wall line cout = 3)

With the default "Fast - 0.2mm" settings
21 min
Outer wall 4min

So if one could take the outer wall movement from "Fine" and the rest from "Fast" the print would theoretically take 30 mins [21 mins - "Fast" outer shell + "Fine" outer walls]  and still have the smooth walls from "Fine", similar to the __"Infill Layer Thickness" setting.

__Would that feature be possible?
How to get around the problem that arise on slopes?
On especially steep slopes?
On bridges?__ 

I asked the almighty internet about this idea, but found nothing really, so I'm asking here now.
I hope my idea is understandable, otherwise ask me for some sketches. 
Unfortunately my programming skills are not sufficient to try it myself.

Thisismydigitalself, 2017-12-15:
I guess this means walls before infill. we need to hear what the smart guys have to say. we don't always see the bigger picture.

mpsdskd, 2017-12-15:
It would also mean Outer wall before Inner wall, but IIRC this is a feature already

Ghostkeeper, 2018-01-02:
I've thought about implementing this sort of thing for walls and skin too. It's certainly possible. Once you have that you can basically get a sort of variable layer height by slicing at 0.01mm height and then combining up to 10 layers together (including the walls). There was the problem of flow changes during the print, but it worked out much nicer with the existing profiles than other solutions for variable layer height. However with VariSlice now being implemented in the recent research sprint it is no longer really necessary for now.

You won't need to explicitly use Walls Before Infill with this sort of thing. I think you're saying that because with a wall of 0.2mm high and infill of 0.1mm high the wall would "stick out" higher than the infill? But you can just wait with printing the wall until the next layer of infill is completed and you won't have any problem with the nozzle hitting that higher wall.

3.2 will feature freeze around the 20th of January (according to our current schedule; things might change still). Shortly after we'll release another beta.

Thisismydigitalself, 2018-01-02:
Looking forward for the release of 3.2. 
from my bad past experience with 3.1Beta, this time  i will force myself to hold on from upgrading my 3.1 since last time i upgraded 3.0.4 to 3.1 beta i literally suffered and it took a complete OS cleanup from any Cura installation before i could install and stabilize 3.1 released version. It will be harder this time since I see lots of new and exiting features are included in 3.2.

ChrisTerBeke, 2018-01-02:
You can easily run 3.1 and 3.2 beta side by side, just don't uninstall the older version. Each version has their own config folder, so the 2 won't mix their settings and screw up that way. The 3.2 beta will copy the 3.1 settings and then make changes to that if needed when opening for the first time or when using that particular version, 3.1 will be untouched by any 3.2 activities.

Thisismydigitalself, 2018-01-02:
These are good news.
I do wonder thou why my upgrade to 3.1beta got messed up and then couldn't install 3.1 released before erasing all folders everywhere.

ChrisTerBeke, 2018-01-02:
3.1 beta and 3.1 used the same data folder (as they are both 3.1). The beta contained a bug that messed up the data storage in some use cases, making any subsequent 3.1 stable install corrupted from the start. Removing the 3.1 folder (and keeping the previous 3.0 folder) and then restarting Cura 3.1 fixed the issue for most cases, but there we a handful of edge cases that we didn't anticipate. Those were manually fixed by emailing the affected users.

We don't expect any of these issues for 3.2 (beta) as there aren't major upgrades in the code that affect the stored user data.

mpsdskd, 2018-01-02:
VariSlice certainly sounds like an interesting feature, thanks for the info.

But it does not solve the Problem of the rough optical and haptic finish, which occurs with big layer heights also on vertical walls (as far as I understood VariSlice would choose the maximum possible layer height on vertical walls)
See: https://s3-eu-west-1.amazonaws.com/3dhubs-knowledgebase/layer-height/photo6.jpg

Even if the calculation with the default print profiles exaggerates the speed difference, there still would be an advantage in the print speed at IMO little to no quality penalties over using the finer print profile.

Ghostkeeper, 2019-02-22:
We've just discussed this but we see some bears on the road for implementing it. Because it'll be quite difficult, the cost vs. gain for this is quite high and we won't be implementing this soon. Some day. Good research topic, maybe.

mpsdskd, 2019-02-22:
This then bears the question what those bears are explicitly? (See what I did there?^^)
Especially as you said you already had some prototype (https://github.com/Ultimaker/Cura/issues/2982#issuecomment-354752739).

Ghostkeeper, 2019-02-22:
We've never made a prototype. The predictions in that comment were purely theoretical.

The three most major roadblocks that we've predicted are:
* It's going to need to change the flow rate (mm³/s) through the nozzle from one wall to the next, which is a lot of flow changing. This is problematic for most materials, especially since the flow changes will be visible on the outside. It is especially problematic for Ultimaker printers which use a Bowden tube.
* Combining walls can only be done in areas where the wall is thick enough. That means that either the walls have to align exactly on top of each other, or you're going to have some gaps somewhere.
* It's going to be difficult to prevent collisions of the nozzle with the printed walls of various heights while not also travelling around all over the place.

megablue, 2019-03-07:
> We've never made a prototype. The predictions in that comment were purely theoretical.
>
> The three most major roadblocks that we've predicted are:
>
> * It's going to need to change the flow rate [...]

#5341, i did some testings, it actually work a lot better than what you worried. you can apply slopes detection similar to how you handle supports,  perhaps, an easy implementation would be just fault a slicing process when the model's slopes angle are more than the user preset. the time saving is huge.

marc-bruenisholz, 2021-03-25:
Hi
I am a student at Lucerne University of Applied Sciences and Arts in Switzerland. I am in a team that is trying to modify CuraEngine to allow printing with multiple, different sized extruders to speed up the printing process, increase stability, but still retain the surface layer quality.
It seems to be possible to skip some layers before another infill layer is printed with the "Infill Layer Thickness" property. There does not seem to be a property for the inner wall layer thickness that can be configured independently from the outer wall layer thickness.
We would like to ask what the state of this feature is. We are grateful for any help we can get.

Ghostkeeper, 2021-04-12:
There is no Outer Wall Layer Thickness either. The major roadblocks are still in place.

Currently this ticket is "deferred", meaning that we think it would improve value for the user, but the amount of work it would take to properly implement it is too great to make it worth our time. There are other things with a greater value per unit of work. So basically we're saying "don't count on it any time soon".

closed 2023-07-13 by github-actions[bot]