Ultimaker/Cura #2983: Printhead Settings, Bounding Box Generation, and Printing in One at a Time Mode

drewoprea, 2017-12-15:
I'm having an issue getting Cura (3.1.0) to properly slice a series of small objects.
I am printing with a MonoPrice Ultimate 3D printer (Wanhao Duplicator 6 clone)
I am starting with the Prusa i3 print settings and adjusting them for my printer.

I load three copies of my object into the build space, set one at a time mode and get the error: "Unable to slice" "Nothing to slice because none of the models fit the build volume"

My Printhead Settings are as follows:
X min 25 mm
Y min 20 mm
X max 65 mm
Y max 50 mm
Gantry Height 55 mm

I have noticed that Cura renders two bounding boxes in One at a Time mode.  One bounding box changes size when I change my print head settings, (I don't think it changes correctly) but one bounding box does not change size.

Included is a screen shot showing the error text, and the two bounding boxes I'm describing.

[image]

nallath, 2017-12-20:
One of the boundingboxes has nothing to do with the head size, but with the certain avoidance settings (eg; Travel distances, platform adhesions, etc).
The second boundingbox has do with one at a time, where in certain cases certain objects can (or can not) be printed.

drewoprea, 2017-12-20:
Ok, in my attached image the smaller box changes shape when i modify the print-head dimension settings, but it is always left/right symmetrical, even if I have asymmetrical dimensions set. Is this some sort of limitation in Cura, or possibly with the Prusa i3 default print settings?
I will test further and look for the other avoidance settings.

drewoprea, 2017-12-21:
So with small enough models (<10mm X,Y,Z) I can put two on the build plate.  It requires that no part of any bounding box overlap, or leave the build area. I can't get four to slice properly. I'll roll my own printer config and report back.  Part of the bounding box appears to be controlled from the setting menu, part from the printer config.

Ghostkeeper, 2018-01-15:
The print head collision area in Cura will always be symmetrical because the objects are re-ordered such that the print head can reach the printed object from the side that is less wide. For instance, if you have a print head where the nozzles are all the way to the front (lots of fans and such behind the nozzles but nothing in front of the nozzles) then it will print the front-most objects first and then move further towards the back, so those fans won't collide with the fans.

For the technically inclined, the actual computation is as follows: We mirror the X and Y axes of the print head shape (180 degree rotation) and take the intersection of that shape and the original print head shape. The result is Minkowski-summed with the model's collision area. The mathematical proof behind this, why this always results in an ordering that can be printed, is quite complex though.

drewoprea, 2018-01-15:
> then move further towards the back, so those fans won't collide with the fans.
so the fans won't collide with the parts?

Thank you for the explanation, I'll keep that in mind while I write the config for my printer.

I still think the math cheats me out of build space, but I can't demonstrate it while i'm using the prusa config on my wanhao.

I think I should be able to print a 3 x 3 grid of small enough parts.  Is slicing this many models (in one are a time mode) something very few people bother to do?

Ghostkeeper, 2018-09-19:
We'll consider all questions answered, then.

closed 2018-09-19 by Ghostkeeper