Ultimaker/Cura #2985: Loading machines takes a long time

peteruithoven, 2017-12-15:
When using the latest 3.1 AppImage or 2.7.0 from the PPA I notice that the "Loading machines..." stap takes a very long time. In 2.7.0 (from ppa) this step takes > 10 seconds, with 3.1 AppImage this takes > 40 seconds. 
I tried starting the 2.7.0 from the terminal and I noticed this line:
```
2017-12-15 21:16:42,202 - DEBUG - UM.Settings.ContainerRegistry.load [214]: Loading data into container registry took 10.63440465927124 seconds
```

I'm running Elementary OS Loki (build on Ubuntu 16.04.3 LTS) on a laptop with a Intel i7-6700HQ (8) @ 3.500GH processor and a SSD harddisk.
Please let me know if I can provide more information.

ChrisTerBeke, 2017-12-16:
3.2 will fix this.

closed 2017-12-16 by ChrisTerBeke

peteruithoven, 2017-12-16:
Awesome! Thanks Chris.