Ultimaker/Cura #2987: PC filament-Bad nozzle temp configuration

Calvin6, 2017-12-15:
Hi,

Im new here and didnt found anything that makes reference to this, so if there is a topic about this I will be gratefull if someone tell my where. Im using a Prusa MK2s printer and printing with Polycarbonate filament. When I set the files to print on the printer, the nozzle automatically sets the temperature to 210°C for the first layer(The default temperature is set to 270°C in the PC profile in Cura). Everything else is fine, the second layer calibrates to 270°C and the bed temperature set to the 107°C or wathever I set, but whenever I want to print with this material i have to change the first layer nozzle temperature manually.

ChrisTerBeke, 2017-12-16:
Related to https://github.com/Ultimaker/Cura/issues/1296?

mareklinka, 2017-12-16:
We are noticing the same thing with the same printer, Prusa MK2S. The issue started showing up after upgrade to 3.1, I'm 99% positive this used to work fine with 3.0.x. I can doublecheck on Monday if necessary.

Calvin6, 2017-12-16:
@ChrisTerBeke I dont think so because I have only 1 extruder and this only occurs me when I use PC configuration. I will check this anyway to rule it out.

@mareklinka Im using 3.1 as well but I didnt print anything in PC before this upgrade, I will try it in an older version and post here if the issue dissapared.

Thanks

Calvin6, 2017-12-19:
Hey,

@mareklinka You were right, in the previous version of Cura this does not happen. I made an update to the firmware of the printer an now the temperature goes to 210° with any Cura profile, hope they fix in the next update.

ChrisTerBeke, 2017-12-19:
In that case it's probably caused by https://github.com/Ultimaker/Cura/issues/1296, which will be fixed in the next release.

closed 2017-12-19 by ChrisTerBeke