Ultimaker/Cura #2988: Feature Request: Alternative to Raft,  Fill First Layer First

damon-atkins, 2017-12-16:
Here is a idea for the Public Domain. It would be nice if their was an option to use brim and tick a box to Fill First Layer, then do the borders against the Fill. I suspect this will help with layer down circles, which seem to pull them selves off the bed if you do not use a raft.

Ghostkeeper, 2018-01-04:
You mean like, close all inner holes in the first layer? Or did you mean to print the skin first, then the walls and then the skirt/brim?

damon-atkins, 2018-01-04:
If a have heaps of holes/circles the shell of the holes are dragged out of shape unless I use a raft which gives them something to stick too.

By printing the fill first it gives the shell for the holes something to connect to.  This way I am hoping that don't need to waste filament on a raft, because the first layer fill is acting like a raft.

The first layer fill is simple back and forth, each line of the fill sticks to the next line. So if the first line goes now well the rest should be fine. Not sure if its worth having a special adjustment for speed. Its important that the nozzle does not left between the first line and the next few, to make sure the lines are not dragged.

So print the Brim first to prime the nozzle, then print the first layer fill, then the fist layer shell, shell will stick to the Brim and the first layer fill. e.g. The holes will not be dragged. Its important the shell connects to the fill first and not printed away from it first and then connects towards the completion of the shell.

It would be useful for prints which are flat on the glass.

Ghostkeeper, 2018-01-04:
There is usually no infill on the first layer. There is skin though, so that's what you probably meant.

There is currently no way to determine the order of skin vs. infill at all. Doing that requires some thought on how that would cooperate with "Infill Before Walls" because ideally you'd be able to order the three arbitrarily...

damon-atkins, 2018-01-04:
New to 3D printing, still trying to learn the terms.

damon-atkins, 2018-06-16:
Any updates on printing Brim -> Skin -> Shell.  Thanks

sleipner451, 2018-07-25:
+1 on this feature request.

I have the same problem: I have a bunch of small holes in the bottom layer, and if the "skin" was printed before the walls of the holes, the wall lines would have something to stick to. Currently, it almost always happen that the tiny circles of the hole walls don't properly stick to the plate and everything is messed up. The feature would be identical to "infill before walls" which unfortunately only works on the layers above the bottom skin.

damon-atkins, 2018-09-28:
You need to thumbs up the original opening comment so it is sorted higher.
[Issues Sorted by Top Thumbs Up](https://github.com/Ultimaker/Cura/issues?q=is%3Aissue+is%3Aopen+sort%3Areactions-%2B1-desc)

damon-atkins, 2018-11-17:
This is slightly related to #3837 `Added experimental settings to print small features slower.`

damon-atkins, 2018-11-17:
#649 `Bottom layer speed should be variable for lines and circles`

Ghostkeeper, 2018-11-20:
Our project manager just removed this from our planning. We won't get time to implement this any time soon.

damon-atkins, 2018-11-21:
Thanks for the frank update.

damon-atkins, 2018-11-26:
Doing more three 3D printing with small holes, it so easy for the circles to be dragged out of position.

I still think this would be a fundamental improvement to the first layer.

@smartavionics has had a go at trying to improve circles, however the advantage of this is its not experimental. Its just an order change for the first layer. 

I wish I could instantly learn the code so I could do this myself for the community.

sleipner451, 2018-11-26:
Just saw this has been removed from planning :( to me it seems like a simple feature, compared to many others, which might make a real difference to small features in the first layer.

Any chance this can be reconsidered?

DDDirk, 2018-11-26:
Because the skin makes 180° turns, if it has nothing to cling to (which would be the circle if that has been printed first), it might get pulled back and not provide the material for the circle to cling to (I have had this happen, but only occasionally (can't remember the settings)). However, the circle will generally be more crucial, so doing it in the proposed order makes sense to me. So maybe the skin needs to be printed very slow, at least close to the turning point (the circle).

jpeterse, 2018-12-08:
Some sort of solution would be really appreciated - especially for large format printers.

The issue I have is, that I can make the small circles stick on the first layer, by adjusting first layer height, slow down print speed, add adhesion with glue stick, or PEI - name you poising.

But with large prints, when the adhesion is cranked up to 11 to make the small circles stick, the resulting large print is next to impossible to remove when finished.
The new flex plate systems resolve this to some extent, but my large format printers have glass beds. and the flex plates in those sizes are really - really expensive.
I always dread having to remove those large prints. Makes me want to break out the sledge hammer from the toolbox.

So some sort of slicer support to solve this, whether the solution is to approve the change to slow down print speed for only small features on first layer, or allow for a setting to print the skin before walls on first layer, would be great.

smartavionics, 2018-12-08:
Hello @jpeterse , it didn't make it into Ultimaker Cura but my Cura variant does have a feature to slow down walls on the first layer

[image]

You can find my Cura for Linux, OS X or Windows at: https://www.dropbox.com/sh/s43vqzmi4d2bqe2/AAADdYdSu9iwcKa0Knqgurm4a?dl=0

jpeterse, 2018-12-08:
@smartavionics Brilliant - I like your approach to the problem.
Might have to give your version a try, but I'd really prefer for it to be included in the stardard release.

jpeterse, 2018-12-09:
@smartavionics
Wow - I tried out your version of Cura. What a difference it makes.

I did a test print of a 210 x 210 mm Vesa mount (just the first layer) with 20 6mm holes scattered throughout.
I print on clean glass with no adhesion assistance, no glue, no PEI, just super clean glass.
And it worked brilliantly.
The walls of the holes stuck down and stayed in place, nice and clean.
The walls of the holes even stuck down so well, that two of them stayed behind, stuck to the glass, when I peeled of the print.

This feature is absolutely brilliant. It only marginally increases the print time. In turn, you are fare more likely to have a successful print.

I know I said I prefer to use the official releases, but I think am converted.
I will continue to use your version for as long as you make it available.

Thank you so much for making this available.

smartavionics, 2018-12-09:
You're welcome.

Almost all of the differences between mb-master Cura and Ultimaker Cura have at some time been offered to Ultimaker for inclusion in their Cura and have either been refused or ignored. So these days I am only contributing bug fixes and not bothering to send them any new features.

smartavionics, 2018-12-09:
I forgot to mention another setting which I don't think is in Ultimaker Cura and that is Initial Layer Wall Speed. I use that most of the time to ensure that my first layer walls stick well.

jpeterse, 2018-12-12:
I don't think I need to slow down the perimeter on the parts I typically print, but I can see it being useful for smaller parts, or parts with smaller sections on them.
I now get so good adhesion, that I was able increase the distance to the print bed a bit.
Now when the print is done, I just leave it on the print bed. I can hear faint creaks as the print bed and the printed part cool at different rate. Then it just pops off on it's own. I have never been able to do that before.

I just don't understand why features like this, that are so helpful and assists in making successful prints are not picked up immediately by the Ultimaker team, while they still spend time and effort on things like spaghetti infill. Especially when the new feature is already developed and proven in the field.
Make one wonder doesn't it?

smartavionics, 2018-12-12:
> I just don't understand why features like this, that are so helpful and assists in making successful prints are not picked up immediately by the Ultimaker team, while they still spend time and [...]
Makes one wonder doesn't it?

They have their own agenda and priorities.

Ellecross, 2019-12-04:
almost one year later .... 

Good news! We discussed this today in the team and we appealed to our overlords! We're going to put this in our backlog (which means that it will be further discussed on how we can implement and then we'll put it in our sprint). yeaaaaaaaaahhhhhh <3 

Sorry we couldn't put it in sooner, guys. But there's hope when we keep pushing. 

UM Cura devs, please refer to CURA-5703

Ghostkeeper, 2019-12-05:
I'll close this as a duplicate of https://github.com/Ultimaker/Cura/issues/6734. Please continue the discussion there.

closed 2019-12-05 by Ghostkeeper