Ultimaker/Cura #2989: Extruder Preheater

azeemiqbal123, 2017-12-16:
There is no option to preheat the extruder. When I click on the print button it starts heating the bed where there is no ability in my printer version of heating the bed. I need an option to directly heating and loading the extruder and starting to print.

fieldOfView, 2017-12-16:
I’ll try adding that in time for Cura 3.2

azeemiqbal123, 2017-12-18:
Please do! It has been a really painful experience trying to go about it. I
had to download 14.0.7 version to make that happen.

Ghostkeeper, 2018-01-04:
Have to be a bit careful with this though because the plastic filament could degrade.

Ghostkeeper, 2018-03-02:
This was removed from our planning by our project manager. It has no priority right now.

fieldOfView, 2018-03-02:
But... there’s a PR for it!

Ghostkeeper, 2018-03-02:
Oh, yes, here it is: https://github.com/Ultimaker/Cura/pull/3144

Seems we had two tickets about it. CURA-4949 is still in there.

diegopradogesto, 2018-07-03:
The PR was merged so I close this issue.

closed 2018-07-03 by diegopradogesto