Ultimaker/Cura #2990: Feature Request: Layer Bed Temperature Adjust

damon-atkins, 2017-12-16:
This idea is for the Public Domain.  Some times its nice to printer hotter bed for the first layer and then allow the bed to cool down a bit to a different temperature.

smartavionics, 2017-12-16:
Err, it can already do that.

damon-atkins, 2017-12-16:
I'll look harder for the option.

damon-atkins, 2017-12-16:
Setting should be under materials of the filament setting, it has bed temp but does not have first layer bed temp (however it is in with all the other settings)

Thisismydigitalself, 2017-12-16:
[image]

damon-atkins, 2017-12-17:
The screen shot is not from where you define the material. Under Preferences -> Materials

[image]

Should be a lot more settings here, and then what you have shown show be just a view or have an option to save it back to the material setting in preferences.

filipgoc, 2017-12-17:
That's a fair point @damon-atkins; there's no user-facing reason not to have 1st layer temp as a part of the fallback material defaults.

Trouble is, the material definition (XML) supports only a limited number of settings - not the full shebang.

I first ran into this with granular fan speed control per material definitions a long time ago, but now the the team already has some improvements in this department on their internal tracker ([mentioned in this issue](https://github.com/Ultimaker/Cura/issues/1135)), so my guess is this could reasonably soon be addressed.

damon-atkins, 2017-12-17:
I assume the idea of Under "Preferences -> Materials" is that you get a new role and do some test prints to work out the settings and the "real thickness" and then create a profile for it.  Maybe even give it a number as well and label the role of filament.  The main interface should just be a viewer, or have adjustments which are saved back to the preference->material settings.  Maybe allow two lines for the description of the material and always show it even if the search does not match. (Maybe section heading should always be shown if not already)

damon-atkins, 2017-12-17:
Closed, handed over to https://github.com/Ultimaker/Cura/issues/1135

closed 2017-12-17 by damon-atkins