Ultimaker/Cura #2991: EGL/GLES Compatibility bug in grid.shader

jwalt, 2017-12-16:
Hi!

Using ARM Mali drivers, Cura's gid.shader does not compile. There are two issues with it.

* One is the same as Ultimaker/Uranium#117 (and the suggested fix works fine)
* The other one is that function ``fwidth`` is missing.  The driver does support ``fwidth``, it just isn't available by default in EGL/GLESv2 and must be explicitly requested

A shader preamble like this works perfectly:

```GLSL
fragment =
    #extension GL_OES_standard_derivatives : enable
    #ifdef GL_ES
        #ifdef GL_FRAGMENT_PRECISION_HIGH
            precision highp float;
        #else
            precision mediump float;
        #endif // GL_FRAGMENT_PRECISION_HIGH
    #endif // GL_ES
```

thopiekar, 2017-12-23:
Could you provide a PR for the second point?

Thanks!

Ghostkeeper, 2018-01-15:
I'd rather put that extra line inside the `#ifdef GL_ES`, since you say that the disabled `fwidth` is only a problem in `EGL/GLESv2`. Maybe you could check if this still works?

    fragment =
        #ifdef GL_ES
            #extension GL_OES_standard_derivatives : enable
            #ifdef GL_FRAGMENT_PRECISION_HIGH
                precision highp float;
            #else
                precision mediump float;
            #endif // GL_FRAGMENT_PRECISION_HIGH
        #endif // GL_ES