Ultimaker/Cura #2992: Long/Abrupt Travels Tearing Up First Layer (v3.1.0)

zembacraftworks, 2017-12-16:
I'm having an issue in v3.1.0 where the first layer will get torn up due to some long, abrupt travels...

After the inner wall, the extruder [image], and this almost always peels up the section of wall that was just laid down. 

This isn't an issue at all in Cura 15, where the infill is started [image]. I haven't checked any of the versions inbetween yet.

Anyway, [image].

I've been digging through the settings looking for something relevant, but so far haven't found anything.

smartavionics, 2017-12-16:
Hi, what it's doing is travelling to a position to start the skin lines such that, if possible (given any hole or weird outlines), it will print the skin in one go rather than starting in the middle and printing one half and then travelling to print the other half.

Personally, I have never seen the behaviour you describe above so I wonder if you can make some adjustment to your retraction/travel speed/bed adhesion that can stop it ripping up the wall it has just extruded.

zembacraftworks, 2017-12-16:
Thanks for the insight. I was troubleshooting bed adhesion for awhile, but it was otherwise great so I don't think it was the main issue.

I was able to get around it by cranking down the Initial Layer Travel Speed (it had been at 50 mm/s).

smartavionics, 2017-12-16:
> I was able to get around it by cranking down the Initial Layer Travel Speed (it had been at 50 mm/s).

Really?, I am using 150 mm/S for travel on my first layer. I don't have adhesion problems and what helped a lot was printing the walls slow (20 mm/S). I am using a modified version of Cura that lets you specify different speeds for the initial layer walls and skin. That way, I can print the walls dead slow to get great adhesion but then the skins get printed quicker which saves time. I did submit the mods to the Cura devs but they didn't want to do that but it works really well for me.
Another thought, do you have combing enabled? It is enabled by default. If so, I recommend that you set it to "no skin" so that it won't try and do that travel move that's causing your problem without doing a retract (assuming you have retraction enabled). I use retract with z-hop.

Ghostkeeper, 2018-01-04:
I have the travel speed on the initial layer way down as well, typically, for this reason.

If it's caused by starting from a better corner in order to reduce the number of seams in the skin, then this would also have been a problem if the rectangles were aligned slightly differently in other models. It's sort of a coincidence.

I don't think it's really a problem. This is why we have settings to tune like the initial layer travel speed.

closed 2018-01-04 by Ghostkeeper