Ultimaker/Cura #2993: Cura 2.3.1 won't start

mlune, 2017-12-16:
I tried rebooting and reinstalling, no effect. Simply fails to start.
here is message it  gives me:
Version: 2.3.1
Platform: Windows-10-10.0.14393
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 61, in <module>
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Application.py", line 310, in getInstance
  File "D:\2.3\build\inst\lib\python3.4\site-packages\cura\CuraApplication.py", line 173, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\Qt\QtApplication.py", line 94, in __init__
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 107, in upgrade
  File "D:\2.3\build\inst\lib\python3.4\site-packages\UM\VersionUpgradeManager.py", line 281, in _upgradeFile
  File "C:\Python34\lib\codecs.py", line 319, in decode
MemoryError

fieldOfView, 2017-12-16:
Go to C:\Users\[you]\AppData\local\cura, and delete cura.log

mlune, 2017-12-17:
WTF, it was that simple. Big thanks

closed 2017-12-17 by mlune