Ultimaker/Cura #2994: Enhancement: An option to make visible all settings with overrides across all sections.

filipgoc, 2017-12-17:
Something like 'Check customized' or  'Check modified' 

[image]

[image]

**Why?** 
It's currently easily possible to make _all_ settings visible (through the settings menu) or hide them, but it is not easily possible to just show all settings with overrides - while that's the most useful view for me.

Basically the same function as the 'i' icon has in each section - but globally, for all sections.

Ghostkeeper, 2018-01-04:
We are planning to use that space to allow the user to select from a number of presets (4 or so), ranging from basic to advanced to expert with more granularity. We'd have to find some different place to put it. Or maybe we could include that in the drop-down or whatever the interface for that will be.

fieldOfView, 2018-01-04:
I have a much better place to put this, for quicker access:
[image]
A hamburger menu next to the setting search field could give quick access to both the setting visibility profiles @Ghostkeeper suggests, as well as the "modified settings" and all settings as a quick toggle.

I made a start with this some time ago (https://github.com/Ultimaker/Cura/compare/master...fieldOfView:feature_setting_visibility_profiles), but was somewhat disappointed with the performance of switching to "all settings" and back and then got caught up in other things.

ChrisTerBeke, 2018-01-04:
The performance should be better after merging the QtQuick Controls 2.0 branch.

fieldOfView, 2018-01-04:
I will revisit my branch when the QtQuick Controls 2 branch has merged.

filipgoc, 2018-01-04:
@fieldOfView I'm loving the setting visibility without going to the main preferences!

If this gets married with Ghostkeeper's note, it will be heavenly.

So, 
instead of just 'normal set'
there would be 'minimalist set, normal set, advanced set, crazy advanced set'

fieldOfView, 2018-01-04:
> one could even be allowed to save custom sets of settings

That's what I started out doing.

fieldOfView, 2018-01-04:
Switching from the default set to All Settings still takes around 5 seconds with the QtQuick Controls2 branch, though it does not lock up the interface. I guess creating the ~300 widgets just takes time.

ChrisTerBeke, 2018-03-02:
A new feature "setting visibility presets" will be added in the next version.

closed 2018-03-02 by ChrisTerBeke