Ultimaker/Cura #2995: Enhancement Idea: Do not hide empty sections

filipgoc, 2017-12-17:
When all settings are hidden, Cura completely hides that section of settings now. 

I think it's more logical to keep the sections visible at all times, and just play with visibility of the items inside. 

One use case for example an attempt to 'reverse' the effect of 'checking all' - as that displays an unholy number of settings.

Currently, if one goes ahead and 'checks none', then all the categories just disappear; making it impossible to go section by section and simply click the 'i' icon for example.

Ghostkeeper, 2018-01-04:
There is the "Defaults" button on that preferences page as well which reverses the effect of checking all.

We hide the empty sections on purpose to give more of a feeling of cleanliness. There's no need to display the "Mesh Fixes" category header for a novice user since it only contains expert settings that are invisible and magic to him.
For an expert user it may be somewhat more convenient. I reset my settings so often during development and every time have to find the Travel category again if I want to test something there. But for a novice user it's just confusing. Especially with that turnstyle that still allows you to collapse and expand a category that is empty.

filipgoc, 2018-01-04:
Makes sense. 

As long as we have better settings visibility toggling, (discussed https://github.com/Ultimaker/Cura/issues/2994#issuecomment-355254131), this is not needed.

closed 2018-01-04 by filipgoc