Ultimaker/Cura #2996: Trying to run 2 machines printing different objects at the same time - usb issues?

swainbow, 2017-12-17:


nallath, 2017-12-20:
Well, running USB from a computer is not really something I'd recommend in the first place and I would definitely not recommend running two USB devices at the same time. 

Every time you start Cura, it tries to autodetect what ports have a printer attached to it. For every printer that it finds, it adds the option to send a print to it.

swainbow, 2017-12-20:
Hi Jaime,
Thank you so much for replying to me. 
I'm aware it's not recommended, but for my work flow it's the best option I've got. 
My problem is that, when I open up 2 screens of cura, it automatically detects both printers on the same screen, and then doesn't say it's connected on the other one? But it worked for a while, and now it doesn't, so that's what's got me confused. 
I was wondering if there was a way to turn off the auto connection function and manually load each printer? 
Thanks for your time,
Derek Swain

nallath, 2017-12-21:
There is no way to do that at the moment. But if you want better USB printer control, have a look at printrun. The USB printing isn't important to us, so it doesn't get a lot of attention (which leads to sort of working features).

Ghostkeeper, 2018-01-15:
> How can I set the printers to only communicate to specific com ports?

That is an open issue. It currently simply picks the first COM port it finds, which depends on your operating system. This is a duplicate of
* https://github.com/Ultimaker/Cura/issues/1049
* https://github.com/Ultimaker/Cura/issues/2267
* https://github.com/Ultimaker/Cura/pull/1806
* https://github.com/Ultimaker/Cura/issues/1244
* https://github.com/Ultimaker/Cura/issues/1777
* https://github.com/Ultimaker/Cura/issues/1434
* https://github.com/Ultimaker/Cura/issues/1321
* https://github.com/Ultimaker/Cura/issues/736

Printing multiple objects at the same time requires being able to connect to multiple printers at the same time, which has a lot of implications for the Monitor stage: Multiple camera streams and associated processing power, multiple clustered printers with their own queues, keeping multiple connections alive, etc. I wouldn't count on it just yet.

fieldOfView, 2018-01-15:
> Printing multiple objects at the same time requires being able to connect to multiple printers at the same time, which has a lot of implications for the Monitor stage: Multiple camera streams and [...]

You are making this sound a lot more complicated than it is, especially since OP is only talking about USB printing. So no camera streams, no multiple clustered printers.

Now that @nallath's printeroutputdevices rewrite is merged, I will have another look at getting manual port selection to work. I got half-way once (https://github.com/Ultimaker/Cura/pull/1688)

nallath, 2018-01-17:
I've taken some extra steps in the architecture to ensure that selecting manual ports should be pretty doable. Didn't have enough time to actually build it, but there is a manual selection shaped hole now :)