Ultimaker/Cura #2997: Open Cura 3.03 Projects with Cura 3.1 ?

filipgoc, 2017-12-17:
Hi, I know there are some issues related to the having installed 3.1 BETA, which I did, but now I am simply trying to open 3.03 projects with 3.1 and it is not working. No 3.1 BETA involved. 

Namely, 3.1 does not load/respect the 3.03 project's setting. 

Is there any known workaround/ fix?

It'd be a pain to try and redo all mu projects from scratch not knowing what transferred and what did not...

**Infill in 3.03: cubic**
[image]

And **infill in 3.1, same project: zig zag**
[image]

Thisismydigitalself, 2017-12-17:
Consider yourself lucky. After i installed 3.1 beta my Cura is a complete mess.  i don't know what broke down but i do know i am in pain. I will not stop using Cura because I love it and the good guys are doing their utmost but I am counting the hours and minutes for next release as the only version i can run now is 3.1beta (no other 3.x version will run anymore) and all my projects needed to be redone and I don't want to think what if i could not reuse these with 3.2..

filipgoc, 2017-12-17:
@Thisismydigitalself uff. sorry to hear that. I'm sure we'll get to the bottom of it sooner or later.

Ghostkeeper, 2018-01-04:
I can't reproduce this. I created 2 projects using 3.0.4, one with the Ultimaker 2+ and one with the Ultimaker 3. Both set the Infill Pattern to Cubic. Then I closed 3.0.4, cleared my settings and loaded 3.1. I could open both projects and both of them would set the Infill Pattern properly to Cubic.

Maybe it would help to try your project file if you are at liberty to share it?

filipgoc, 2018-01-04:
@Ghostkeeper thanks for checking. I'm attaching a project file made for IMADE3D JellyBOX and I have verified the issue does show up on my computer using this specific project file. 

- Sidenote, is the only difference between 3.03 and 3.04 the name (Cura vs Ultimaker Cura)? I created the projects with 3.03. 
- You also made me think about the printer definitions. Maybe the issue is the printer definitions do not match - if a project file is opened with a different printer (like custom), it does all sorts of unpredictable things.  Even though I did no change them in months...

[door handle.3mf.zip](https://github.com/Ultimaker/Cura/files/1603450/door.handle.3mf.zip)

[image]
[image]

Ghostkeeper, 2018-01-22:
I think I remember 3.0.4 had a fix for the signing of the MacOS build as well. But that's not relevant for Windows users.

Indeed your project file specifies that it should be cubic, but loading it into version 3.2 causes the zigzag pattern to be selected. Looks like a version upgrade problem between 3.0 and 3.1 due to that extruder train for single extrusion machines refactor we did. That should really have been tested better.

filipgoc, 2018-01-22:
Thanks for looking into this @Ghostkeeper.

fieldOfView, 2018-01-22:
snazzy logos you have there, @filipgoc

filipgoc, 2018-01-22:
> snazzy logos

:-) Haha. It was the fastest and least cognitively taxing way to make sure I am actually opening the project files in the correct Cura versions as I was trying to prove that one of them has issues.

Ghostkeeper, 2018-01-29:
Our testers report that it's fixed for 3.2 stable.

closed 2018-01-29 by Ghostkeeper