Ultimaker/Cura #2998: Cura 3.1 View shows full support, but printer (ultimaker 3) only pringts half support height

OhmegaStar, 2017-12-17:
Hello,

I've upgraded to Cura 3.1, after the upgrade, when i slice a model with lots of support in it, and send it to the printer (Um3 lates 4.x firmware) the supports are only printed half way up the structure, even if in layers mode it seems that the planned supports are all the way up.

I'm at a loss how to troubleshoot this.

(attaching gcode file)
[UM32_62-mm-jar-holder-heavy-duty.zip](https://github.com/Ultimaker/Cura/files/1565752/UM32_62-mm-jar-holder-heavy-duty.zip)

As for now I guess i will have to install an older version so my prints will work again.

Ghostkeeper, 2018-01-04:
The g-code looks all right. You can also load it into a different application that can load g-code, such as RepetierHost. It'll show that the support is complete and that the line widths are correct.

Are you sure it wasn't just that the feeder started grinding or something?

OhmegaStar, 2018-01-04:
Hi, 

It may have been grinding (am using PLA supports), but the problem came with this particular model when i changed the slicer/generator to cura 3.1, and went away once i downgraded to the previous version (cura 2.7 i think it was)

Best Regards,

Henrik

Ghostkeeper, 2018-01-22:
Maybe you could take a look at what the difference between the two g-codes is then.

I looked at it again and it seems it would print fine. I'll start a test print.

Ghostkeeper, 2018-01-22:
In my test with that g-code it got lots of overextrusion. Maybe you should reduce the support infill density.

OhmegaStar, 2018-01-26:
Hi Ghostkeeper, 

Thanks for looking into this.
I have no experience in reading gcode, so I would probably miss the important parts.

The Support infill density is by design (i use PVA - not PLA - Support) in order to ensure a level roof.

Ill try to install cura 3.1 again, and reprint. it will be a couple of days before i have time to do so.

I'll also have a closer look at the support pattern and infill properties, i need a level roof, but if i can make it with less support then that would be very good.

Best Regards,

Henrik

closed 2018-03-02 by ChrisTerBeke