Ultimaker/Cura #3000: cura won't slice (not auto, not manual)

mikee0070, 2017-12-17:
tried all versions  until version 2 slices after version 2 dos not slice any projects
dont know what happened was working file a couple of months ago ...

LOGS:
2017-12-17 18:15:17,632 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FileLogger
2017-12-17 18:15:17,679 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin FirmwareUpdateChecker
2017-12-17 18:15:17,679 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeProfileReader
2017-12-17 18:15:17,695 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeReader
2017-12-17 18:15:17,695 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin GCodeWriter
2017-12-17 18:15:17,695 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ImageReader
2017-12-17 18:15:17,711 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LegacyProfileReader
2017-12-17 18:15:17,711 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin LocalFileOutputDevice
2017-12-17 18:15:17,726 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MachineSettingsAction
2017-12-17 18:15:17,757 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin OctoPrintPlugin
2017-12-17 18:15:17,757 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PerObjectSettingsTool
2017-12-17 18:15:17,773 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PluginBrowser
2017-12-17 18:15:17,773 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin PostProcessingPlugin
2017-12-17 18:15:17,773 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RemovableDriveOutputDevice
2017-12-17 18:15:17,789 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimulationView
2017-12-17 18:15:17,789 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SliceInfoPlugin
2017-12-17 18:15:17,804 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SolidView
2017-12-17 18:15:17,804 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin CameraTool
2017-12-17 18:15:17,820 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin MirrorTool
2017-12-17 18:15:17,820 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin RotateTool
2017-12-17 18:15:17,820 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin ScaleTool
2017-12-17 18:15:17,836 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SelectionTool
2017-12-17 18:15:17,836 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin TranslateTool
2017-12-17 18:15:17,851 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UltimakerMachineActions
2017-12-17 18:15:17,882 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UM3NetworkPrinting
2017-12-17 18:15:17,882 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UpdateChecker
2017-12-17 18:15:17,898 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin USBPrinting
2017-12-17 18:15:17,914 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin UserAgreementPlugin
2017-12-17 18:15:17,914 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade21to22
2017-12-17 18:15:17,929 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade22to24
2017-12-17 18:15:17,929 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade25to26
2017-12-17 18:15:17,929 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade26to27
2017-12-17 18:15:17,945 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade27to30
2017-12-17 18:15:17,945 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin VersionUpgrade30to31
2017-12-17 18:15:17,945 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin SimpleView
2017-12-17 18:15:17,961 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin X3DReader
2017-12-17 18:15:17,961 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XmlMaterialProfile
2017-12-17 18:15:17,976 - INFO - UM.PluginRegistry.loadPlugin [238]: Loaded plugin XRayView
2017-12-17 18:15:17,976 - INFO - UM.VersionUpgradeManager.upgrade [112]: Looking for old configuration files to upgrade.
2017-12-17 18:15:17,992 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-12-17 18:15:17,992 - ERROR - UM.Backend.Backend.startEngine [84]: Unable to find backend executable: D:\Cura3.1\CuraEngine.exe
2017-12-17 18:15:24,574 - INFO - cura.Settings.GlobalStack.addExtruder [74]: Extruder[a8 fdmextruder] added to [a8] at position [0]
2017-12-17 18:15:24,574 - WARNING - UM.Settings.ContainerStack.addMetaDataEntry [162]: Meta data with key machine was already added.
2017-12-17 18:15:24,574 - DEBUG - UM.Settings.ContainerRegistry.load [231]: Loading data into container registry took 6.441695213317871 seconds
2017-12-17 18:15:24,965 - DEBUG - UM.Controller.setActiveView [86]: Setting active view to SolidView
2017-12-17 18:15:24,980 - WARNING - cura.Settings.GlobalStack.addExtruder [70]: Extruder [a8 fdmextruder] has already been added to this stack [a8]
2017-12-17 18:15:24,980 - WARNING - UM.Settings.ContainerStack.addMetaDataEntry [162]: Meta data with key machine was already added.
2017-12-17 18:15:24,980 - WARNING - cura.Settings.ExtruderManager.registerExtruder [220]: Extruder [a8 fdmextruder] has already been registered on machine [a8], not doing anything
2017-12-17 18:15:25,371 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromVertices [420]: Calculating normals took 0.06250977516174316 seconds
2017-12-17 18:15:25,371 - DEBUG - STLReader.STLReader.read [79]: Loaded a mesh with 8052 vertices
2017-12-17 18:15:25,387 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [387]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.0 seconds. 36 input vertices. 36 output vertices.
2017-12-17 18:15:27,325 - DEBUG - UM.Qt.Bindings.Theme.load [164]: Loading theme file: D:\Cura3.1\resources\themes\cura-light\theme.json
2017-12-17 18:15:27,403 - DEBUG - UM.Qt.Bindings.Theme.load [232]: Loaded theme D:\Cura3.1\resources\themes\cura-light
2017-12-17 18:15:27,543 - WARNING - cura.Settings.GlobalStack.addExtruder [70]: Extruder [a8 fdmextruder] has already been added to this stack [a8]
2017-12-17 18:15:27,543 - WARNING - UM.Settings.ContainerStack.addMetaDataEntry [162]: Meta data with key machine was already added.
2017-12-17 18:15:27,543 - WARNING - cura.Settings.ExtruderManager.registerExtruder [220]: Extruder [a8 fdmextruder] has already been registered on machine [a8], not doing anything
2017-12-17 18:15:27,700 - DEBUG - UM.Controller.setActiveView [86]: Setting active view to SolidView
2017-12-17 18:15:28,481 - DEBUG - OctoPrintPlugin.DiscoverOctoPrintAction._createAdditionalComponentsView [270]: Creating additional ui components for OctoPrint-connected printers.
2017-12-17 18:15:28,512 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action._createAdditionalComponentsView [138]: Creating additional ui components for UM3.
2017-12-17 18:15:28,590 - DEBUG - UM3NetworkPrinting.DiscoverUM3Action.startDiscovery [38]: Starting printer discovery.
2017-12-17 18:15:29,171 - DEBUG - UM.Qt.QtRenderer._initialize [182]: Support for Vertex Array Objects: True
2017-12-17 18:15:29,177 - DEBUG - UM.View.GL.OpenGL.__init__ [92]: Initialized OpenGL subsystems.
2017-12-17 18:15:29,178 - DEBUG - UM.View.GL.OpenGL.__init__ [93]: OpenGL Version:  4.1.0 NVIDIA 388.59
2017-12-17 18:15:29,180 - DEBUG - UM.View.GL.OpenGL.__init__ [94]: OpenGL Vendor:   NVIDIA Corporation
2017-12-17 18:15:29,181 - DEBUG - UM.View.GL.OpenGL.__init__ [95]: OpenGL Renderer: GeForce GTX 1080/PCIe/SSE2
2017-12-17 18:15:29,182 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura3.1\resources\shaders\default.shader]...
2017-12-17 18:15:29,185 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura3.1\resources\shaders\selection.shader]...
2017-12-17 18:15:29,187 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura3.1\resources\shaders\default.shader]...
2017-12-17 18:15:29,190 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura3.1\resources\shaders\composite.shader]...
2017-12-17 18:15:29,193 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura3.1\resources\shaders\overhang.shader]...
2017-12-17 18:15:29,195 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura3.1\resources\shaders\striped.shader]...
2017-12-17 18:15:29,201 - DEBUG - UM.View.GL.ShaderProgram.load [55]: Loading shader file [D:\Cura3.1\resources\shaders\default.shader]...
[... cut ...]
2017-12-17 18:15:33,973 - DEBUG - UM.Backend.Backend._createSocket [214]: Previous socket existed. Closing that first.
2017-12-17 18:15:34,051 - DEBUG - UM.Backend.Backend._logSocketState [174]: Socket state changed to Listening
2017-12-17 18:15:34,066 - ERROR - UM.Backend.Backend.startEngine [84]: Unable to find backend executable: D:\Cura3.1\CuraEngine.exe
2017-12-17 18:15:34,356 - DEBUG - CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [348]: Sending slice message took 0.38378047943115234 seconds

mikee0070, 2017-12-17:
and yes the file D:\Cura3.1\CuraEngine.exe is there and works if i open it with cmd ...

nallath, 2017-12-19:
Do you have permissions to run CuraEngine? You could try running Cura with Admin priveleges.

fieldOfView, 2017-12-19:
> You could try running Cura with Admin priveleges.

Bad Nallath... BAD NALLATH!

fieldOfView, 2017-12-19:
@mikee0070: Just for troubleshooting purposes, could you try installing Cura to its default location in C:\Program Files?

mikee0070, 2017-12-19:
tried with admin privileges - same problem
tried install to defaul location, even on C:\ - same problem

benphelps, 2017-12-20:
I was slicing a model with no issues and then everything quit, checked the log and I'm now getting this error as well. 

I have no anti-virus running other than Windows Defender and it's clean.

Here are some logs and my windows version is `Microsoft Windows 10 Pro 10.0.16299 Build 16299`: https://gist.github.com/benphelps/241031863f5d8e0316d823bc55e1f555

I can run the CuraEngine.exe from the command line just fine:
```C:\Users\ben>"C:\Program Files\Ultimaker Cura 3.1\CuraEngine.exe"

Cura_SteamEngine version DEV
Copyright (C) 2017 Ultimaker```

I  tried running Cura as administrator and that also failed.

nallath, 2017-12-20:
Whaaat? I just wanted to know if for some magical reason the curaengine needed admin priveleges. I've seen that happen before when people messed around with parts of the system.

benphelps, 2017-12-21:
I copied the engine and required DLLs to a new folder on my desktop and set the backend location in my cura.cfg and it still gives the same error.

nallath, 2017-12-21:
Well, for some magical and unknown reason, it can't find the executable. I don't have the foggiest why this would fail, especially since you seem to be the only one that has this issue.

I did improve the logging on this part, as it's rather sparse (probably because it's old code and it never actually failed before...

mikee0070, 2017-12-21:
im not the only one .., but there a so few of us with the problem  that there wont be a fix i guess

benphelps, 2017-12-21:
There's 2 people here and at least 3 other issues on this same problem.

#1555 was on this issue, wasn't a virus scanner, was just dismissed
#2283 is also this same exact issue
#1114 again is on this issue

LipuFei, 2017-12-21:
@benphelps When that problem shows up, could you check in Task Manager see if there is a `CuraEngine.exe` process running?

benphelps, 2017-12-22:
There is no CuraEngine.exe process running.

benphelps, 2017-12-23:
I tried adding Cura to my PATH and it didn't work.

Ghostkeeper, 2018-01-15:
I still can't reproduce this. It's hard to tell for us what's going on. We can only guess at an answer (admin privileges, antivirus, etc). I hope that someone figures out a workaround eventually and maybe the workaround allows us to find a real solution eventually.

ianpaschal, 2018-03-26:
@mikee0070 @benphelps 
Can you guys verify if running newer versions solves this problem for you?

mikee0070, 2018-03-26:
hi
so win10 cura 3.2.1 i open up a stl file it says slicing... and thats all so no fix yet...
tried - full uninstalls did not help...
works fine on my other win10 pc

ianpaschal, 2018-03-26:
That's unfortunate but it also means there's a very high chance that there's something wrong with your specific computer. Without your computer in front of us it will be nearly impossible to reproduce and subsequently fix...

mikee0070, 2018-03-26:
yeah but as i said i am not the only one with this problem...

benphelps, 2018-03-26:
I eventually discovered a core issue with my Windows installation that caused other programs to exhibit very bizarre behavior as well. It was missing the Null (NUL) device (which acts essentially as /dev/null). This was causing other programs to fail to open or simply crash with seemingly random "file was not found" esq. errors as well.  You can find some other instances of this issue with a Google search for "[windows null device missing](https://www.google.com/search?q=windows+null+device+missing)".

A simple check for this problem is an attempt to write to the Null device, from a command prompt run `echo test > nul`.  On a system with no issues, you'll get no output, on a system with this issue, you'll get `The system cannot find the file specified.`.

I was never able to find a solution to this problem (finding the problem was hard enough) and finally used the Windows 10 restore feature to do a fresh reinstall.

I'm pretty confident in this being the cause of my problems here, this almost always showed up when programs interacted with other CLI tools (Git, OpenVPN and Python also had issues for me) and attempted to redirect the output to the NUL device.

tituszx, 2018-04-02:
I had same issue when firewall block connection.
With unblocked connection in firewall Cura slice model with and without internet connection.

Ghostkeeper, 2018-04-26:
As far as I know, we never write to any NUL device in CuraEngine. One of our dependencies might though.

dsimonet, 2018-09-02:
Same here. By default, any application can't access to internet on my computer. Can't make any slicing if firewall's blocked but it work like a charm went authorized ! I was about to think slice was performed on external server ^^

ianpaschal, 2018-09-03:
@dsimonet I wish. Then we'd have a great way to speed up slicing time by 600% if we switched to distributing Cura Engine with Cura and slicing it locally. 🤔 😛

dsimonet, 2018-09-03:
haha !! Yes good point. For the moment this is absolutely not efficient. Because slicing not starting at all ! I don't have any problem to let Cura have internet access. But it took me 5 mn to figure it out.

Ghostkeeper, 2018-11-05:
We still don't really know how to reproduce this. It seems very system-dependent.

dsimonet, 2018-11-05:
You can use SimpleWall from Henry++ to prevent Cura from accessing internet. Normally after this Cura will can't slice anymore.

Ghostkeeper, 2019-02-19:
Does your SimpleWall also prevent accessing local socket connections (not on the internet, but just a local socket)? Because if so, it makes sense that Cura is not able to find CuraEngine. The two processes communicate via a local socket.

mikee0070, 2019-02-19:
cura 3.6.0 works again :)

closed 2019-02-19 by mikee0070