Ultimaker/Cura #3001: Cura 3.1 is not launching on Win 7

labrador21, 2017-12-17:
Hello,

I'am trying to install Cura 3.1 on a laptop with windows 7(64bit). Although the installation process finishes, cura can not be started.
I already installed the new update "Cura-3.2.0-master.20171211145115-win64.exe" from #2916 but this doesn't work, too.

**The following report.wer-file can be found on the computer after the installation process:**

Version=1
EventType=APPCRASH
EventTime=131580212030874823
ReportType=2
Consent=1
UploadTime=131580212031484858
ReportIdentifier=b30beec3-e374-11e7-a203-c80aa9954bf4
IntegratorReportIdentifier=b30beec2-e374-11e7-a203-c80aa9954bf4
Response.BucketId=604453047
Response.BucketTable=28
Response.type=4
Sig[0].Name=Anwendungsname
Sig[0].Value=Cura.exe
Sig[1].Name=Anwendungsversion
Sig[1].Value=0.0.0.0
Sig[2].Name=Anwendungszeitstempel
Sig[2].Value=59ad373a
Sig[3].Name=Fehlermodulname
Sig[3].Value=atio6axx.dll
Sig[4].Name=Fehlermodulversion
Sig[4].Value=6.14.10.9704
Sig[5].Name=Fehlermodulzeitstempel
Sig[5].Value=4b8ddfd7
Sig[6].Name=Ausnahmecode
Sig[6].Value=c0000005
Sig[7].Name=Ausnahmeoffset
Sig[7].Value=0000000000d28ffb
DynamicSig[1].Name=Betriebsystemversion
DynamicSig[1].Value=6.1.7601.2.1.0.768.3
DynamicSig[2].Name=Gebietsschema-ID
DynamicSig[2].Value=1031
DynamicSig[22].Name=Zusatzinformation 1
DynamicSig[22].Value=6199
DynamicSig[23].Name=Zusatzinformation 2
DynamicSig[23].Value=619904affd4215a865c62e683071933d
DynamicSig[24].Name=Zusatzinformation 3
DynamicSig[24].Value=8bd8
DynamicSig[25].Name=Zusatzinformation 4
DynamicSig[25].Value=8bd87424258359e26be90bd5ce4bc1d2
UI[2]=C:\Program Files\Ultimaker Cura 3.2\Cura.exe
UI[3]=Cura.exe funktioniert nicht mehr
UI[4]=Windows kann online nach einer Lösung für das Problem suchen.
UI[5]=Online nach einer Lösung suchen und das Programm schließen
UI[6]=Später online nach einer Lösung suchen und das Programm schließen
UI[7]=Programm schließen
LoadedModule[0]=C:\Program Files\Ultimaker Cura 3.2\Cura.exe
LoadedModule[1]=C:\Windows\SYSTEM32\ntdll.dll
LoadedModule[2]=C:\Windows\system32\kernel32.dll
LoadedModule[3]=C:\Windows\system32\KERNELBASE.dll
LoadedModule[4]=C:\Windows\system32\SHLWAPI.dll
LoadedModule[5]=C:\Windows\system32\GDI32.dll
LoadedModule[6]=C:\Windows\system32\USER32.dll
LoadedModule[7]=C:\Windows\system32\LPK.dll
LoadedModule[8]=C:\Windows\system32\USP10.dll
LoadedModule[9]=C:\Windows\system32\msvcrt.dll
LoadedModule[10]=C:\Program Files\Ultimaker Cura 3.2\python35.dll
LoadedModule[11]=C:\Windows\system32\WS2_32.dll
LoadedModule[12]=C:\Windows\system32\RPCRT4.dll
LoadedModule[13]=C:\Windows\system32\NSI.dll
LoadedModule[14]=C:\Windows\system32\ADVAPI32.dll
LoadedModule[15]=C:\Windows\SYSTEM32\sechost.dll
LoadedModule[16]=C:\Windows\system32\VCRUNTIME140.dll
LoadedModule[17]=C:\Windows\system32\api-ms-win-crt-runtime-l1-1-0.dll
[... 106 lines cut ...]
AppPath=C:\Program Files\Ultimaker Cura 3.2\Cura.exe

**And this stderr.log-file can be found in the cura-folder:**

Fatal Python error: Segmentation fault

Current thread 0x00001f2c (most recent call first):
  File "X:\master\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGLContext.py", line 29 in setContext
  File "X:\master\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGLContext.py", line 104 in detectBestOpenGLVersion
  File "X:\master\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 77 in __init__
  File "X:\master\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 212 in __init__
  File "X:\master\build\inst\lib\python3.5\site-packages\UM\Application.py", line 330 in getInstance
  File "X:\master\build\inst\bin\cura_app.py", line 85 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

**Any help is appreciated!**

LipuFei, 2017-12-17:
Hi @labrador21 that "Cura-3.2.0-master.20171211145115-win64.exe" from #2916 is not a stable version and it doesn't contain a working fix for network storage on Windows.

Could you Remove this unstable version and post the log file you get with 3.1? Thanks!

And please also remove the following folders if they exist, otherwise when a future 3.2 release actually arrives, there might be problems with that.
`C:\Users\<your-username>\AppData\Roaming\cura\3.2`
`C:\Users\<your-username>\AppData\Roaming\cura\master`
`C:\Users\<your-username>\AppData\Local\cura\3.2`
`C:\Users\<your-username>\AppData\Local\cura\master`

labrador21, 2017-12-18:
Hi LipuFei,

thanks for your quick response.
I uninstalled the "Cura-3.2.0-master.20171211145115-win64.exe" and  removed all files you mentioned above. After that I installed Cura 3.1.0. While the installation window was shown, the programm stayed quite a long time at the following position:

**Starte: "C:\Program Files\Ultimaker Cura 3.1\vcredist_x64.exe" /install /quiet /norestart**

Allthough the finished window was shown after a while, cura didn't start at all.
I found the following files after the installation on my laptop:

**Report.wer**
Version=1
EventType=APPCRASH
EventTime=131580957144638142
ReportType=2
Consent=1
UploadTime=131580957145328182
ReportIdentifier=2f3e8900-e422-11e7-a28d-c80aa9954bf4
IntegratorReportIdentifier=2f3e88ff-e422-11e7-a28d-c80aa9954bf4
Response.BucketId=604453047
Response.BucketTable=28
Response.type=4
Sig[0].Name=Anwendungsname
Sig[0].Value=Cura.exe
Sig[1].Name=Anwendungsversion
Sig[1].Value=0.0.0.0
Sig[2].Name=Anwendungszeitstempel
Sig[2].Value=59ad373a
Sig[3].Name=Fehlermodulname
Sig[3].Value=atio6axx.dll
Sig[4].Name=Fehlermodulversion
Sig[4].Value=6.14.10.9704
Sig[5].Name=Fehlermodulzeitstempel
Sig[5].Value=4b8ddfd7
Sig[6].Name=Ausnahmecode
Sig[6].Value=c0000005
Sig[7].Name=Ausnahmeoffset
Sig[7].Value=0000000000d28ffb
DynamicSig[1].Name=Betriebsystemversion
DynamicSig[1].Value=6.1.7601.2.1.0.768.3
DynamicSig[2].Name=Gebietsschema-ID
DynamicSig[2].Value=1031
DynamicSig[22].Name=Zusatzinformation 1
DynamicSig[22].Value=6199
DynamicSig[23].Name=Zusatzinformation 2
DynamicSig[23].Value=619904affd4215a865c62e683071933d
DynamicSig[24].Name=Zusatzinformation 3
DynamicSig[24].Value=ab74
DynamicSig[25].Name=Zusatzinformation 4
DynamicSig[25].Value=ab749ae48f27a682ed705821884e2a92
UI[2]=C:\Program Files\Ultimaker Cura 3.1\Cura.exe
UI[3]=Cura.exe funktioniert nicht mehr
UI[4]=Windows kann online nach einer Lösung für das Problem suchen.
UI[5]=Online nach einer Lösung suchen und das Programm schließen
UI[6]=Später online nach einer Lösung suchen und das Programm schließen
UI[7]=Programm schließen
LoadedModule[0]=C:\Program Files\Ultimaker Cura 3.1\Cura.exe
LoadedModule[1]=C:\Windows\SYSTEM32\ntdll.dll
LoadedModule[2]=C:\Windows\system32\kernel32.dll
LoadedModule[3]=C:\Windows\system32\KERNELBASE.dll
LoadedModule[4]=C:\Windows\system32\SHLWAPI.dll
LoadedModule[5]=C:\Windows\system32\GDI32.dll
LoadedModule[6]=C:\Windows\system32\USER32.dll
LoadedModule[7]=C:\Windows\system32\LPK.dll
LoadedModule[8]=C:\Windows\system32\USP10.dll
LoadedModule[9]=C:\Windows\system32\msvcrt.dll
LoadedModule[10]=C:\Program Files\Ultimaker Cura 3.1\python35.dll
LoadedModule[11]=C:\Windows\system32\WS2_32.dll
LoadedModule[12]=C:\Windows\system32\RPCRT4.dll
LoadedModule[13]=C:\Windows\system32\NSI.dll
[... 108 lines cut ...]
ConsentKey=APPCRASH
AppName=Cura.exe
AppPath=C:\Program Files\Ultimaker Cura 3.1\Cura.exe

**stderr.log**
Fatal Python error: Segmentation fault
Current thread 0x000015ec (most recent call first):
  File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGLContext.py", line 29 in setContext
  File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\View\GL\OpenGLContext.py", line 104 in detectBestOpenGLVersion
  File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Qt\QtApplication.py", line 73 in __init__
  File "X:\3.1\build\inst\lib\python3.5\site-packages\cura\CuraApplication.py", line 209 in __init__
  File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Application.py", line 323 in getInstance
  File "X:\3.1\build\inst\bin\cura_app.py", line 82 in <module>
  File "Console.py", line 24 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load
  File "__startup__.py", line 12 in <module>
  File "<frozen importlib._bootstrap>", line 634 in _load_backward_compatible
  File "<frozen importlib._bootstrap>", line 664 in _load_unlocked
  File "<frozen importlib._bootstrap>", line 958 in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 969 in _find_and_load

**Thanks for your help in advance!**

LipuFei, 2017-12-19:
Hi @labrador21 , I'm not sure what happened, but it could be related to the graphics driver. Could you upgrade your graphics driver(s) to the latest version(s) and try again? Thanks

labrador21, 2017-12-22:
Hi LipuFei, although I had the latest graphics driver installed, I deleted my grahic hardware and reinstalled it again. Nevertheless I couldn't start cura 3.1. After that I tried the different versions of cura. I had to go down to cura 2.5. This programm runs on my labtop. The main difference I recogniced is, that this version installs "visual studio 2010 redistibutable" and all newer programms install "visual studio 2015 redistibutable". Maybe this could be the reasion why cura 2.6 and all newer versions do not work on my computer.

labrador21, 2017-12-25:
Hi @LipuFei  , after several trials with the newest drivers from my graphic card manufacturer with no results, I installed the latest drivers from the web page of my laptop manufacturer. Although this driver was older,  I now can start curo 3.1.
Many thanks for your hint!

closed 2018-01-04 by jackha