Ultimaker/Cura #3004: [question] Creating and sharing profiles

ChrisRiddell, 2017-12-18:
Hello,

I have a question if I want to make a material for eg: PETG for UM2+ that has its own set of profiles is it best to make the files like these

- resouces
  - materials
    - colorfabb_petg.xml.fdm_material
  - quality
    - ultimaker2_plus
      - um2p_cf_petg_0.4_fast.inst.cfg
      - um2p_cf_petg_0.4_normal.inst.cfg

and in the inst.cfg files ref the material using the metadata tag 

`material = colorfabb_petge_ultimaker2_plus_0.4_mm`

if this is the right way of going about this is their any doc on what all the metadata vars do for eg looking at the ones already used their is 

`weight = -1` or `weight = 1`

What does weight mean as I noticed some of the quality profiles use a -1 and a 1

also for the version and setting_version what does that mean as they ones already their are set to a 4 is this just the version of the profile so Id use a 1?

fieldOfView, 2017-12-18:
The "weight" metadata is only used for sorting the profile in lists.
The "setting_version" is used for compatibility with certain version of Cura. The current version of Cura requires the setting_version to be 4.

ChrisRiddell, 2017-12-18:
if i have a couple colours 

eg 

colorfabb_petg_white.xml.fdm_material
colorfabb_petg_black.xml.fdm_material

How do I tell the quality files to be available to both what would I need to set my `material`` as as when I just had the one I done `material = colorfabb_petgeconomy_ultimaker2_plus_0.4_mm`

nallath, 2017-12-20:
Cura has some fallback strategies. As you might have noticed, all our quality profiles refer to generic materials. If you have a specific material type selected (say blue PLA), it will first try to find a quality based on blue PLA. As there is no quality for blue PLA, it will try to find one for generic PLA, and use that.

ChrisRiddell, 2018-01-19:
I have anther question is their an easier way to package and import these type of profiles that have a materiel and related quality files? as the only way I have found is getting the user to copy the files to the correct location in the C drive or add it to the OSX app package.

fieldOfView, 2018-01-19:
You don’t have to add the files to the application folder, you can also drop them in the configuration folder. Having said that, a .curaresources package similar to the .curaplugin would be nice.

ChrisRiddell, 2018-01-19:
configuration folder?

ChrisRiddell, 2018-01-19:
Do I keep the directory structure? eg: resouces/** or do I put the materials and quality files just in AppData\Roaming\cura\3.1

ChrisTerBeke, 2018-01-19:
We're thinking a lot currently about how to improve material/profile packages, resource paths and plugins that provide resources. I expect more details soon. We'll take these suggestions into consideration as well.

@Appesteijn input for us :)

ChrisRiddell, 2018-01-21:
Is their a list of Keys that are available to use in the xml.fdm_material?

ChrisTerBeke, 2018-01-21:
https://github.com/Ultimaker/Cura/blob/master/plugins/XmlMaterialProfile/XmlMaterialProfile.py#L943

ChrisRiddell, 2018-02-09:
Anther suggestion for a .curaresources file is to make it possible not to just have material/profiles but also have optional printer def as well that would make it very powerful so printer manufacturers can make a full package for Cura if they choose not to submit it to be apart of Cura.

fieldOfView, 2018-02-09:
... and themes, and postprocessing scripts

ChrisRiddell, 2018-02-28:
Going to close this as my question has been answered and I'm guessing their is already internal talk about a curaresources package file.

closed 2018-02-28 by ChrisRiddell

ianpaschal, 2018-02-28:
Thanks for helping keep things tidy @ChrisRiddell. There is indeed on-going internal discussions about how this sort of thing might work.