Ultimaker/Cura #3012: When failing to open a project via keyboard, cura has to be restarted (two issues)

filipgoc, 2017-12-18:
Application Version: Cura 3.03, Cura 3.1, and probably older, I remember this has been around for a long time. 
 

Platform: Mac Sierra
 

Display Driver:

Steps to Reproduce:

1. open
1. select a project file
1. confirm opening with 'enter' or with a mouseclick
1. in this confirmation project opening dialog, if one clicks on `open` with a mouse, all works fine. 
1. IF one wishes to use keyboard, one issue is that no default button is pre-selected. Normally, in this situation, I would presume the 'open' to be pre-selected as a default choice (after all I am trying to open a project). 
    - still, if I use 'tab' to select open and confirm with spacebar, projects opens fine
1. IF one simply presses 'enter', which on Mac confirms the default pre-selected button as the normal UX, the confirmation video just closes (probably since no button was highlighted) and the project does not open. Yet, what is worse, Cura now has to be restarted to be able to open any projects anew. 

Actual Results:
The project does not open, and Cura has to be restarted to be able to open any projects anew. 

Expected results:

1. the open button should be highlighted by default. this will solve BOTH issues. 

Additional Information:
[image]

Vandrasc, 2017-12-21:
Confirmed! 
Thank you for reporting it, btw this also occurs on other platforms. 

Devs 4735

ianpaschal, 2018-08-01:
This was fixed a long time ago but the GitHub ticket seems to be forgotten.

closed 2018-08-01 by ianpaschal