Ultimaker/Cura #3013: Profile settings not pulling info correctly

tdub415, 2017-12-19:
Application Version: 3.1

Platform: Windows 10

Display Driver: GeForce 388.59

Steps to Reproduce:
Loaded the monoprice magic number profiles from https://downloads.monoprice.com/files/software/21711_Software_171026.zip into the %appdata%cura\3.1\quality folder.  

Actual Results:
When I go to activate the profiles in cura, the layer settings are not matching what they are supposed to be, for example the M3 profile should be .13125 layer height, but it's truncating to .1, the M2 should be .0875 but it pulls up as .6

Expected results:
The profiles should be pulling the layer height as set in their files.

Additional Information:
I have tried using the Setting > Profile > update with current settings/override, but this doesn't actually save/update the profiles.

tdub415, 2017-12-19:
I've also noticed (though not sure if it matters) that I cannot select a nozzle size in Cura 3.1 like I can in Cura 3.0.4

geekcubed, 2017-12-20:
Also having similar issues with 3.1 and layer heights relating to the MonoPrice Mini.  Pulled a Cura3 profile from the MPSM wiki ( https://www.mpselectmini.com/slicers/cura? ), and although it states it's buggy with 3.1, I couldn't of predicted what a massive pain it was going to be.

After about 5 hours of tinkering I finally found chaning `"has_variants"` to `false,` was the only way to get Cura to load the Definition without crashing.  But even after all that, it turns out not all of the included Quality settings work.

Layer heights of **0.13125** and **0.04375** both either cause Cura to fail to slice (with the Layer Height field showing Red) or just crash completely.

```2017-12-20 17:24:15,575 - CRITICAL - cura.CrashHandler.__init__ [62]: An uncaught exception has occurred!
2017-12-20 17:24:15,578 - CRITICAL - cura.CrashHandler.__init__ [65]: TypeError: unable to convert a Python 'str' object to a C++ 'double' instance
2017-12-20 17:24:15,723 - DEBUG - UM.Logger.logException [76]: Exception: An exception occurred in inherit function =0.2 + resolveOrValue('layer_height')
2017-12-20 17:24:15,724 - DEBUG - UM.Logger.logException [80]: Traceback (most recent call last):
2017-12-20 17:24:15,726 - DEBUG - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Settings\SettingFunction.py", line 84, in __call__
2017-12-20 17:24:15,728 - DEBUG - UM.Logger.logException [80]:   File "<UM.Settings.SettingFunction (0x1870de29588) =0.2 + resolveOrValue('layer_height') >", line 1, in <module>
2017-12-20 17:24:15,729 - DEBUG - UM.Logger.logException [80]: TypeError: unsupported operand type(s) for +: 'float' and 'str'
2017-12-20 17:24:15,734 - WARNING - UM.Logger.logException [76]: Exception: Could not validate setting layer_height, an exception was raised
2017-12-20 17:24:15,736 - WARNING - UM.Logger.logException [80]: Traceback (most recent call last):
2017-12-20 17:24:15,737 - WARNING - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Settings\Validator.py", line 64, in __call__
2017-12-20 17:24:15,739 - WARNING - UM.Logger.logException [80]: TypeError: unorderable types: str() < float()
2017-12-20 17:24:16,014 - DEBUG - UM.Logger.logException [76]: Exception: An exception occurred in inherit function =0.2 + resolveOrValue('layer_height')
2017-12-20 17:24:16,017 - DEBUG - UM.Logger.logException [80]: Traceback (most recent call last):
2017-12-20 17:24:16,019 - DEBUG - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Settings\SettingFunction.py", line 84, in __call__
2017-12-20 17:24:16,021 - DEBUG - UM.Logger.logException [80]:   File "<UM.Settings.SettingFunction (0x1870de29588) =0.2 + resolveOrValue('layer_height') >", line 1, in <module>
2017-12-20 17:24:16,023 - DEBUG - UM.Logger.logException [80]: TypeError: unsupported operand type(s) for +: 'float' and 'str'
2017-12-20 17:24:16,030 - WARNING - UM.Logger.logException [76]: Exception: Could not validate setting layer_height, an exception was raised
2017-12-20 17:24:16,032 - WARNING - UM.Logger.logException [80]: Traceback (most recent call last):
2017-12-20 17:24:16,034 - WARNING - UM.Logger.logException [80]:   File "X:\3.1\build\inst\lib\python3.5\site-packages\UM\Settings\Validator.py", line 64, in __call__
2017-12-20 17:24:16,036 - WARNING - UM.Logger.logException [80]: TypeError: unorderable types: str() < float()
```

1. Unhandled exceptions are not cool. Especially when the application presents a couple of empty dialog boxes then quits to desktop.
2. Is there any information out there relating to what changed in the Definition files between 3.1 and 3.0.x? 
3. What's the score with these layer_height values in particular?  All the layers in the linked zip are based off the same multiple, with the same number of decimals.  What makes these two break?

As someone who is just starting out in the world of 3d printing, this has proven to be massively frustrating first experience of Cura.

tdub415, 2017-12-20:
@geekcubed Are you using the printer definition file from monoprice, or the fixed one from https://github.com/Ultimaker/Cura/issues/2968, there was an issue in the print file, not sure if that may be causing issues with the magic number quailty settings as well, I hadn't gotten them to work yet at the point I opened the previous ticket regarding chaning layer heights.

goebish, 2017-12-20:
I'm using the fixed .json but I'm having the same issue with 3.1
So ... I installed 3.0.4 instead, that's better but still the nozzle size dropbox is empty, I'mnot sure if this is normal, that's my first experience with a slicer as well, I should have my first printer (MP select mini v1) by tomorrow ...

geekcubed, 2017-12-20:
@tdub415 I've just seen #2968 and have tinkered a bit more with the Definition based on that:

Set `has_variants`  to false (Line 12) and completely removed the `layer_height` values (Line 40).  I've only go the out-the-box hot end + nozzle, so it's not an issue not be able to customise that particular aspect.

Having done that, I now get rounding errors on quite a few of the Quality settings from that profile zip ("M1 - M8").

 - M1 - 0.04375 rounds to 0.0437 (should be rounding up, but at least it lets me slice with it now)
 - M2 - 0.00875 shows as 0.06
 - M3 - 0.13125 shows as 0.1
 - M4 and M5 show correctly
 - M6 - 0.2625 shows as 0.2
 - M7 and M8 are okay

tdub415, 2017-12-20:
Yep, the rounding errors is exactly why I opened this ticket, and it looks like you're having the same issues with the profiles as me (from what I can recall).  I've also noticed that I'm getting orange and red boxes in the layer height fields on some of the profiles, including the ones that pull the layer heights correctly.

ChrisTerBeke, 2017-12-21:
We have notified MonoPrice about this, they need to make profile changes.

geekcubed, 2017-12-21:
Agreed @ChrisTerBeke that the profile they currently list for Cura3 is broken, but it seems to perhaps be masking another issue?  Given a quality file like : 

```
[general]
version = 2
name = M2 Quality
definition = monoprice_select_mini

[metadata]
type = quality
weight = 1
quality_type = high
setting_version = 3

[values]
layer_height = 0.0875
wall_thickness = 1.05
top_bottom_thickness = 0.72
infill_sparse_density = 22
speed_print = 50
speed_layer_0 = =round(speed_print * 30 / 50)
speed_topbottom = 20
cool_min_layer_time = 5
cool_min_speed = 10
```

Why does the main UI in Cura show the a Layer Height of 0.06?  If you go through to Manage Profiles, it shows the values from the file correctly (0.0875).  How can we tell which value is being used to generate the slices?

ChrisTerBeke, 2017-12-21:
The quality type in the profile is high, which for us means 0.06. Actually we 'restrict' a quality type to a certain layer height in the profiles (not in code, but as a mental model).

goebish, 2017-12-21:
Thanks @ChrisTerBeke , I modified the .cfg files and it seems fine now.
I also modified the .json profile to allow 0.04375 layer height.
https://www.dropbox.com/s/a8hoyt6tkotousp/MP%20Select%20Mini%203.1.zip?dl=1

Ghostkeeper, 2017-12-30:
> 0.04375 rounds to 0.0437

This is probably a floating point rounding error. It thinks you entered 0.0437499992549419403076171875 and then rounds it to 4 decimals.

thetobby, 2017-12-31:
I am experiencing similar issue with material_diameter of 1.75 getting rounded to 1.7 in Cura. Unsure where the value is read from as I have it both in printer definition and quality profile. Files to reproduce are in #3057

Ghostkeeper, 2018-01-17:
Eventually we might have to remove quality_type altogether and just match on layer height, or else rename the quality_types to something containing the layer heights. It's an unreasonable limitation that you have to add new quality types for every adjustment of the layer height beyond the default ones. Many printers actually want different layer heights because of their step size in the Z axis.

closed 2018-01-17 by ChrisTerBeke