Ultimaker/Cura #3014: Cura 3.1 dual extruder nozzle offset not working

schneehorn, 2017-12-19:
Since Cura 3.1 it is not possible to set the Nozzle offsets for dual extruder systems independently. Whenever you change a field in the "Extruder 1" setup, the same value is automatically transferred to the "Extruder 2" setup and vice versa. I.e. I have to define a Nozzle offset of -16.5 mm for Extruder 1 and +16.5 mm for Extruder 2. But changing one value, automatically changes the corresponding value of the other extruder automatically. This holds both, for the Nozzle X offset, as well as the Nozzle Y offset and for the Nozzle size. Setting the offsets independently works fine in Cura 2.7 and Cura 3.04, only issue is Cura 3.1.

Application Version:
Cura 3.1

Platform:
macOS 10.13.2

Steps to Reproduce:
Go to Preferences -> Printers -> Add New Printer
Select Number of Extruder -> 2
Go to Extruder 1 Tab, enter some value in Nozzle offset X
Go to Extruder 2 Tab, same value will show up. Change value, go back to Extruder 1 Tab, same new value shows up....

Vandrasc, 2017-12-19:
Hi @schneehorn, we already fixed this issue and it will be present in the next release. 
Question: Did you used 3.1 Beta before downloading the latest 3.1? If so, please first remove your Beta configuration folder before using the latest release. 

@ChrisTerBeke, do we have any workaround for this? Thank you!

ChrisTerBeke, 2017-12-19:
Unfortunately there is no work around in 3.1, but indeed it'll be fixed in 3.2 (beta), which is due in January.

ChrisTerBeke, 2017-12-19:
Also, duplicate of https://github.com/Ultimaker/Cura/issues/2963.

closed 2017-12-19 by ChrisTerBeke

schneehorn, 2017-12-19:
No, I did not use the beta. Upgraded from 3.0.4 to 3.1 directly. Happy to hear that a fix is coming. Thanks.