Ultimaker/Cura #3016: holes in model are filled when printing

dadriaen, 2017-12-19:
Application Version:
3.1.0

Platform:
Windows 10

Display Driver:
NVIDIA Quadro K2100M

Steps to Reproduce:
upload model that includes some holes (to put screws through). Holes are visible in solid view but are filled in Layer view.

Actual Results:
prints have their holes filled (matching the layer view but not the solid view)

Expected results:
that prints have the same result as the input files

Additional Information:
Display in solid view: [image]

Display in layer view: 
[image]

smartavionics, 2017-12-19:
Hi, could you please make the model available so I can try and emulate your problem?

dadriaen, 2017-12-19:
Dear Mark,

Thanks for the very quick reply! Here you have a link to download the CURA project file: https://www.dropbox.com/s/395zukiqisbzqm3/Model%20171107%203%20stuks.3mf.curaproject.3mf?dl=0.

Thanks!
Dominique

[cid:image005.png@01D378C3.FDABFFE0]
Prof. Dr. Dominique Adriaens
Chair Educational Board Biology
tel: +32 9 264.52.19, fax: +32 9 264.53.44
E-mail: dominique.adriaens@UGent.be<outbind://5-000000002B873A8E03CAD411AE9E005004995AACC4103900/dominique.adriaens@rug.ac.be>

Ghent University – Department of Biology
Evolutionary Morphology of Vertebrates & Zoology Museum
K.L. Ledeganckstraat 35, B-9000 Gent
BELGIUM
Office location: click here<https://soleway.ugent.be/routes/1104>

http://www.fun-morph.ugent.be/
http://www.zoologymuseum.ugent.be/

[cobranding_final]

smartavionics, 2017-12-19:
I loaded a part of your model into FreeCAD which has an analyser and it isn't happy, I think maybe you need to mend your model?

[image]

dadriaen, 2017-12-19:
Thanks, Mark. Does FreeCAD give hints at what level the model needs mending (or does it include a solver)?
Dominique

[cid:image005.png@01D378C5.8EFA11C0]
Prof. Dr. Dominique Adriaens
Chair Educational Board Biology
tel: +32 9 264.52.19, fax: +32 9 264.53.44
E-mail: dominique.adriaens@UGent.be<outbind://5-000000002B873A8E03CAD411AE9E005004995AACC4103900/dominique.adriaens@rug.ac.be>

Ghent University – Department of Biology
Evolutionary Morphology of Vertebrates & Zoology Museum
K.L. Ledeganckstraat 35, B-9000 Gent
BELGIUM
Office location: click here<https://soleway.ugent.be/routes/1104>

http://www.fun-morph.ugent.be/
http://www.zoologymuseum.ugent.be/

[cobranding_final]

[Image removed by sender. screenshot_2017-12-19_11-28-11]<https://user-images.githubusercontent.com/585618/34155072-df6085a8-e4af-11e7-9dd1-4c5814571a51.png>

smartavionics, 2017-12-19:
Hi, sorry, I don't really know that much about it but it has those buttons on the right that can fix the reported problems and then you can export the repaired mesh. There are also other tools on the internet that are regular used for mending meshes.

dadriaen, 2017-12-19:
OK, thanks! I’ll try the solver in freecad and see how it works out. Was trying Meshmixer, but crashes all the time.
Many thanks!
Dominique

[cid:image005.png@01D378C6.F48040E0]
Prof. Dr. Dominique Adriaens
Chair Educational Board Biology
tel: +32 9 264.52.19, fax: +32 9 264.53.44
E-mail: dominique.adriaens@UGent.be<outbind://5-000000002B873A8E03CAD411AE9E005004995AACC4103900/dominique.adriaens@rug.ac.be>

Ghent University – Department of Biology
Evolutionary Morphology of Vertebrates & Zoology Museum
K.L. Ledeganckstraat 35, B-9000 Gent
BELGIUM
Office location: click here<https://soleway.ugent.be/routes/1104>

http://www.fun-morph.ugent.be/
http://www.zoologymuseum.ugent.be/

[cobranding_final]

dadriaen, 2017-12-19:
Thanks, Mark! Got the mesh issues solved in FreeCad, en now it looks perfect in the layer view in Cura.
Best
Dominique

[cid:image005.png@01D378CA.D6A9DE60]
Prof. Dr. Dominique Adriaens
Chair Educational Board Biology
tel: +32 9 264.52.19, fax: +32 9 264.53.44
E-mail: dominique.adriaens@UGent.be<outbind://5-000000002B873A8E03CAD411AE9E005004995AACC4103900/dominique.adriaens@rug.ac.be>

Ghent University – Department of Biology
Evolutionary Morphology of Vertebrates & Zoology Museum
K.L. Ledeganckstraat 35, B-9000 Gent
BELGIUM
Office location: click here<https://soleway.ugent.be/routes/1104>

http://www.fun-morph.ugent.be/
http://www.zoologymuseum.ugent.be/

[cobranding_final]

smartavionics, 2017-12-19:
That's great. If you are now happy, please close this issue. Cheers.

closed 2017-12-19 by dadriaen