Ultimaker/Cura #3018: Cura Projects - Not saving printer profile

Thisismydigitalself, 2017-12-19:
Application Version:
 Cura 3.1Beta
Platform:
 win10/64
 

Display Driver:
Nvidia 388.13
Steps to Reproduce:
configure 2 printers
Actual Results:
save project with a given printer. 
Expected results:
even if cura is now on a different printer profile, when i load a project it should load the printer it was saved with. it is not. project appears to not save the printer profile.
Additional Information:

ChrisTerBeke, 2017-12-19:
I know you had issues with 3.1 stable, so I'm not gonna ask to upgrade, but I'm pretty sure this was fixed in the stable (and next release as well).

ChrisTerBeke, 2017-12-19:
Btw, you selected "load project", not "import models" I assume?

Thisismydigitalself, 2017-12-19:
Yes, unfortunately i can not run any other version now.
I save and open *.3mf files.

ChrisTerBeke, 2017-12-19:
When importing a 3MF, there is also an option to only import the models. Maybe that one is accidentally enabled in preferences instead of loading the whole project?

Thisismydigitalself, 2017-12-19:
Thank you. you were right. 
just changed  that in general settings and it works just fine.
hope all other settings are imported as saved. will check. if all's well, you can close this issue.

closed 2017-12-19 by ChrisTerBeke