Ultimaker/Cura #3019: Dual extruder - one nozzle temp question

karabas, 2017-12-19:
When  start printing last layer last extruder Cura turns off temp of the previous extruder. And the print is failed. 
Cura does not know about such a setup.  Logically this behaviouring must be set in end gcode.
So howto stop Cura turning off the temp before print ends?

ChrisTerBeke, 2017-12-19:
Can you provide some information? When creating a new issue there's a template you can fill out. For this particular issue it would also help to know which printer you're using and attach a project file so we can use that to debug.

Ghostkeeper, 2018-01-04:
That behaviour of turning off the temperature is indeed in the end g-code. You can modify the end g-code by going to the Machine Manager and clicking on "Machine Settings". There you'll find a text box that contains the end g-code where the temperature is set to zero. You should be able to remove those lines of g-code there.

MB3Y, 2018-01-14:
I had filed the same issue earlier. Setting temp to 0 is done by Cura, not in end gcode.
See: [Ultimaker/Cura] Extruder gets cooled down before end of print (#2918)

closed 2018-02-26 by ianpaschal

Ghostkeeper, 2018-03-16:
Let's file this as a duplicate of #2918 then.