Ultimaker/Cura #3020: Cura 3.1Beta: Saving project will not change loaded session file name

Thisismydigitalself, 2017-12-20:
Application Version:
Cura 3.1 Beta
Platform:

Win10/64
Qt:

PyQt:

Display Driver:
Nvidia 388.x
Steps to Reproduce:

Saving a project will not change the name of the loaded session and when sending to octopi for print the name of the project being printed is the name of the file which was loaded to cura  in the first place (before resaving it) - not the re-saved name. without saving i would say this is OK, but saving with a new name must be the new name which will be transmitted to Octopi.
Actual Results:

even though a project name was changed and saved, Cura wont change the name of the file and the old name is transmitted to octopi. the end result is printing a different named project while the same project has different name saved on local disk.
Expected results:

Saving a project (I'm guessing) should change the name in Cura.
Additional Information:

Not sure if this is how it should work but when i save a project i assumed Cura should reflect that both in the GUI and the file being sent to OctoPi.

Vandrasc, 2017-12-20:
Hi, if I understood correctly, you expect that the print job name, which you renamed when saving the project, to automatically reflect in Cura? For example, after loading a file into Cura, the print job name in the left bottom corner will be <machineprefix-filename>, then saving the project file and giving a different name than the automatically given one, I will call it TEST, will you expect to automatically change in Cura? So, now my printjob name is TEST after saving and renaming the project. Correct? 

Currently, Cura will change the name when you will re-load the saved project into Cura, but not when you save and rename the project, this is the intended behavior. If you reload the project and Cura will still display the previous name than the given one, then it is a bug.
Also, the printjob name can be directly changed in Cura interface by just clicking on the printjob name if you wish. 
The file that will be sent to Octoprint will have the same name as displayed in Cura.

Please let us know if this clears up your questions, thank you!

Thisismydigitalself, 2017-12-20:
Hi there.
Having the same project/file with different names on different hosts should not be allowed.
My experience tells me this: if you open a project in Cura, modify it, re-save it and then upload to octoprint - you have to have Cura GUI, Local Host file name, and remote host (in my case octoprint) to have the very same file with its new same name. Right now this is not how it works for me.

Vandrasc, 2017-12-20:
Hi @Thisismydigitalself,

The generated printjob/file name in Cura will be reflected by the last loaded file and the machine prefix. Sorry, we do not support that (yet) but feel free to add a feature request. 
As workaround, reloading the project file will not be necessary, you can change the name in the right bottom corner where the printjob name is being displayed before sending it to Octoprint.

@fieldOfView , what kind of behavior would you expect when sending a file to Octoprint (in this case)? Thank you

fieldOfView, 2017-12-20:
I would expect the filename in OctoPrint to reflect the job name that is used/shown in Cura.

Thisismydigitalself, 2017-12-20:
Shouldn't Cura update its current job name during saving a project with a different name?

fieldOfView, 2017-12-20:
I would not expect the "document" to change by saving it. The job name is part of the document you are creating with Cura.

Vandrasc, 2017-12-21:
@Thisismydigitalself, as this is currently the expected result I am afraid we will not interpret the behavior as a bug, please submit your suggestion through a feature request, we are always eager to improve the user experience. 

As work-around, change the printjob name in Cura to match with your project file name before sending it to OctoPrint. Thank you for your feedback! 

@fieldOfView, thank you!

Thisismydigitalself, 2017-12-22:
Never implied this might be a bug - all i said was that this leads to mistakes and messy databases.
if someone changes a parameter and then saves-as the project before sending to printer - the Gcode that is sent to printer will be different than the one that will be regenerated from saved project.

And, I Don't know how and where to ask for feature requests.

Vandrasc, 2017-12-22:
Oh, sorry! The title and description suggested that it was a bug, my bad. 

_And, I Don't know how and where to ask for feature requests._   Done :)

Ghostkeeper, 2022-03-14:
This was finally implemented in Cura 4.13. If you save a project file, the saved file name will be used (minus extension) as the project name from then on.

closed 2022-03-14 by Ghostkeeper