Ultimaker/Cura #3021: Cura 3.1 Beta & Ultimaker Preview - Network leak

AJanosch, 2017-12-20:
**Application Version:**
3.1.0 at least the Beta (sent to me by Lipu Fei on 12.12.2017)
Ultimaker 3 Firmware: 4.1.4.20171204

**Platform:**
Windows 10

**Steps to Reproduce:**
Open Cura 
Click "Monitor Tab"
Click Ultimaker "Preview" icon - to see the "video of the printing process
Click "Prepare" Tab

**Actual Results:**
Every time you do this will increase network traffic for about 10MBit(/s
see Windows "Task Manager"

**Expected results:**
Repeating this for 6-7 times is driving network Traffic to > 60MBit/s

[Cura_Ultimaker_network_Leak.zip](https://github.com/Ultimaker/Cura/files/1575647/Cura_Ultimaker_network_Leak.zip)

Then the preview Window is not displaying the actual video from Ultimaker any more

**Additional Information:**
See screenshot (63.8 MBit/s is generated by 1 Ultimaker 3 & Cura only!!!!)
[image]

or video 
[Cura_Ultimaker_network_Leak.zip](https://github.com/Ultimaker/Cura/files/1575661/Cura_Ultimaker_network_Leak.zip)

alekseisasin, 2017-12-20:
@AJanosch , Thank you for the information. This issue was solved before, but seems something went wrong.

nallath, 2017-12-21:
It's fixed in the rework. The stream for the webcam wasn't closed correctly in all cases, so it would constantly append an extra one.

AJanosch, 2017-12-22:
if You need a Beta-Tester - please let me know where to find the next Beta version.

nicolinux, 2018-01-11:
Ich can "confirm" this issue in the stable version 3.1.0. As soon as I select the monitor tab with an ongoing print, my ssh session start lagging and typing is slow. Since I already have a shitty internet connection, I didn't catch that at first :)

ChrisTerBeke, 2018-01-11:
As indicated by @nallath, the networking had a complete rewrite. This rewrite won't be in the upcoming 3.2 release (as that's due before testing is completed), but in 3.3. When we've confirmed that the rewrite does not contain the leak we'll close this issue.

closed 2018-01-17 by ChrisTerBeke

AJanosch, 2018-02-20:
Dear  Cura & Ultimaker Team,

Another  comment just from the user perspective:
This is a serious bug. Please put real effort info a fix really soon,

Four more than a half day  about 80 employees in our company were affecting the network performance.
On my colleagues PC generates > 500 MBits/s !!! traffic.
Today, Our system administrator it took >4hrs of localizing the problem!
 (>500MBit/s of traffic with all effects of  DDOS attack caused by Cura & Ultimakers!!! )

The fix was easy: Closing Cura 3.2.1 at his desk !!!!!!!!!
But how many companies are out there, affected by this misbehaviour, without finding the real cause!!!!

ianpaschal, 2018-02-20:
@AJanosch As @ChrisTerBeke said, it _is_ fixed in the next version. Real effort has been applied. 😉

fieldOfView, 2018-02-21:
To elaborate on that: the issue was fixed in a rewrite of the networkprinting functionality. That rewrite was not stable enough in time for Cura 3.2, so it will be part of Cura 3.3.