Ultimaker/Cura #3022: Object tools not allowing simple floats

BagelOrb, 2017-12-20:
Application Version:
Cura 3.1

Platform:
Linux Mint

Qt:
Don't recall how to find this out

PyQt:
Same

Display Driver:
NVIDIA corg(?)

Steps to Reproduce:
- Select object
- Select scale tool or translate tool
- slect a box with a number
- select all
(- optional: delete text)
- Try typing ".123"

Actual Results:
Dot before any digit is ignored, so the end number will be "123" rather than "0.123"

Expected results:
Starting a floating point number with a dot is allowed, so we end up with ".123" or "0.123"

-----------------------
PS: nice issue template! Maybe add some steps for people to find out how to find the info we want them to give. Als we should definitely ask for a .curaproject.3mf file in the additional info

ChrisTerBeke, 2017-12-21:
Mm can't reproduce this on master branch (Windows), it results in 0.123.

p.s. just added some info about 3MF files and where to find the logs to the issue template

BagelOrb, 2017-12-21:
If it doesn't reproduce on windows then never mind. This probably has a very low priority then.

nallath, 2017-12-21:
Also doesn't reproduce for me on Fedora.

Ghostkeeper, 2018-01-15:
Not on Ubuntu 17.10 either, neither from source or with AppImage.

closed 2018-01-15 by Ghostkeeper