Ultimaker/Cura #3023: 4055 Segmentation fault Cura-3.1.0.AppImage on Fedora release 27

juckendesAuge, 2017-12-20:
**Application Version:**

./Cura-3.1.0.AppImage

**Platform:**

uname -a
Linux DESKTOP-Fedora-Linux.fritz.box 4.14.6-300.fc27.x86_64 #1 SMP Thu Dec 14 15:31:24 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

lspci |grep -i vga
01:00.0 VGA compatible controller: NVIDIA Corporation GM204 [GeForce GTX 970] (rev a1)

cat /etc/fedora-release 
Fedora release 27 (Twenty Seven) 64 Bit

**Bug:**
./Cura-3.1.0.AppImage 
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number
Fontconfig warning: "/etc/fonts/conf.d/65-0-lohit-bengali.conf", line 32: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-sans.conf", line 24: unknown element "langset"
Fontconfig warning: "/etc/fonts/conf.d/69-gnu-free-serif.conf", line 24: unknown element "langset"
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
QObject::connect: Parentheses expected, signal MainWindow::
file:///tmp/.mount_P1Fb7u/usr/bin/resources/themes/cura-light/styles.qml:197:17: QML Item: Binding loop detected for property "width"
qml: TableViewSelection: index out of range
qml: TableViewSelection: index out of range
./bin//cura.sh: Zeile 12:  4055 Segmentation fault      (Speicherabzug geschrieben) cura $@

**Steps to Reproduce:**
start the app
./Cura-3.1.0.AppImage

lordwimsey, 2017-12-21:
Hello, I am quite new to github and not yet 100% confident with etiquette, so fill me in, when this is not wanted. I am posting this, because I experience a very similar problem on Linux Mint 18.3.
(At least, I think, it is similar.)

`peter@peter-MS-7713 ~ $ uname -a
Linux peter-MS-7713 4.4.0-21-generic #37-Ubuntu SMP Mon Apr 18 18:33:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
`
`peter@peter-MS-7713 /etc $ cat os-release 
NAME="Linux Mint"
VERSION="18 (Sarah)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 18"
VERSION_ID="18"
HOME_URL="http://www.linuxmint.com/"
SUPPORT_URL="http://forums.linuxmint.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/linuxmint/"
UBUNTU_CODENAME=xenial`

Starting the AppImage from the GUI (doubleclick on downloaded File), I can see the first splash screen with the roboter for like 10 seconds. Then it disappears with no message and Cura won't start. No process resides ("sudo ps -ef | grep cura" shows none other than grep)

Trying to start the AppImage via CLI, I get:
`peter@peter-MS-7713 ~/Downloads/Cura_Alle_Versionen_AppImage $ ./Cura-3.1.0.AppImage
QApplication: invalid style override passed, ignoring it.
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig error: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 72: non-double matrix element
Fontconfig warning: "/etc/fonts/conf.d/10-scale-bitmap-fonts.conf", line 80: saw unknown, expected number`

(Same behaviour as above)

However, when I start Cura as sudo, it starts up and works, but I would rather not having to do this.
The AppImage itself is owned by peter (the only user):
`peter@peter-MS-7713 ~/Downloads/Cura_Alle_Versionen_AppImage $ ls -l
insgesamt 108356
-rwxr-xr-x 1 peter peter 110952448 Dez 21 11:24 Cura-3.1.0.AppImage`

LipuFei, 2017-12-21:
I think this is solved by https://github.com/Ultimaker/Cura/pull/2903

lordwimsey, 2017-12-21:
> I think this is solved by #2903

Please note that I have what seems to be the same issue on Linux Mint, running radeon graphics adapter.

lspci:
`VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Caicos [Radeon HD 6450/7450/8450 / R5 230 OEM]`

LipuFei, 2017-12-21:
Hi @lordwimsey, that fix doesn't distinguish between NVIDIA and AMD cards, so it might work for you. We don't have a machine with AMD graphics card. I have created an AppImage based on 3.1 with that fix. Could you test this for us? Thanks in advance. It's too big to upload here...

https://send.firefox.com/download/fd61d5ec59/#j7-f49ScwOxzK-sQFhh-rA

lordwimsey, 2017-12-21:
Thanks, I am currently downloading that file and will try it tonight. 
I will report results in about an hour.

I really appreciate your effort to help me. Thank you very much.

lordwimsey, 2017-12-21:
I downloaded your file and extracted it. 
Then I started the AppImage by doubleclicking from the gui.
A system message appeared, what I would like to do, and I chose "make executable and run" (my Linux Mint is german, so the english text might differ).
The splash screen appears and cura starts up. 

BUT:
the "add printer" dialog, or all dialogs lack their content, and the "prepare" screen shows no platform grid. 
I tried to upload an image to illustrate, hope this works:
[img]http://up.picr.de/31299461hu.jpg[/img]

Thanks again, this feels like progress... ;-)

LipuFei, 2017-12-21:
@lordwimsey Thanks for your quick reply! I tested the same AppImage on ubuntu and it worked fine. Looks like that OpenGL driver fix worked for you. The empty dialog is something else. Could you upload your log file? It's here: `~/.local/share/cura/3.1/cura.log`.

If you don't mind doing a clean start, you can remove the following folders and run Cura again:
- `~/.local/share/cura`
- `~/.config/cura`
- `~/.cache/cura`

lordwimsey, 2017-12-21:
Ok, the "clean start" (I renamed the folders) did the trick. 
Thanks a lot! Cura starts and works fine now. (Quite laggy, It takes a few seconds to mark some digits with the mouse so I can overwrite them).

I attached the logfile from _before_ the clean start, in case this is of interest.

Again, thanks a bunch, and have nice holidays!
[cura_pre-clean-start.log](https://github.com/Ultimaker/Cura/files/1580911/cura_pre-clean-start.log)

closed 2017-12-22 by LipuFei