Ultimaker/Cura #3024: Request: Allow Extra Skin Wall Count to be negative

whosawhatsis, 2017-12-21:
I'm not sure what the logic is behind the Extra Skin Wall Count setting being set to 1 by default, but it seems like a bad idea to me. I can see how it would be helpful to have a perimeter line separating sparse infill from solid (skin) infill when they occur in the same layer, but where the skin meets the actual (vertical-ish) wall of the print, I think you're just making it harder for the solid infill (skin) lines to be anchored at the ends.

What would make more sense is a value of -1. That is, if your normal walls are made up of two perimeters, "skin" areas would have only one, and the ends of the lines of solid fill would bond directly to the "extra" perimeter on adjacent sparse infill layers.

When sparse and solid infill occur in the same layer, this strategy would also put a perimeter line between the two, but that perimeter would be around the infill part rather than the skin part.

ChrisTerBeke, 2017-12-21:
Wouldn't 0 achieve the same result. You can already set it to 0 right now. As for the defaults, those come out of our materials/testing departments, they always find the best settings for Ultimaker machines.

Ghostkeeper, 2018-01-15:
The anchoring of the skin line will happen to the extra skin wall instead of the normal wall, with correct overlap and such. And the skin wall will be printed just alongside the normal walls like all other walls, so I wouldn't expect the anchoring to be worse.

However there is some merit to have fewer walls where the skin meets the walls than where the infill meets the walls, because the skin has 100% density and as such doesn't "shine through" the walls as the infill does. Only when you create these extra walls for infill to reduce the infill shining through, I'm of the opinion that you might as well have those walls around the skin as well (which slightly reduces print time at usually no other cost) so you might as well just set the number of normal walls one higher.

Also I'd rather create a setting called "Extra Infill Wall Count" than to allow negative values for "Extra Skin Wall Count"  though. I think that would be more intuitive.

whosawhatsis, 2018-01-15:
Extra Infill Wall Count might be more intuitive, but it seems to be increasing the number of settings unnecessarily. I think a better solution would be to have a "skin wall count" and a "infill wall count" setting, both of which (by default) are hidden and inherit from the "wall count" setting, but can be overridden. This would at least be much more consistent with the way other settings are laid out in Cura 2/3

Ghostkeeper, 2018-03-26:
Our project manager deferred this. We aren't planning to implement this soon.

GregValiant, 2024-11-14:
Hello.
Current versions of Cura (5.80 and up) can do this with a modifier mesh and Top Surface Skin Layers.
Can this be closed?
[image]

closed 2024-12-10 by github-actions[bot]